frankarobotics/franka_ros2 #5: `joint_effort_trajectory_controllers` package is probably obsolete

destogl, 2022-05-05:
Hi,

Friendly ros2_control maintainer here :)  Recently we merged [support for effort interfaces into `joint_trajectory_controller`](https://github.com/ros-controls/ros2_controllers/pull/225) from [ros2_controllers](https://github.com/ros-controls/ros2_controllers) repository. I think and (hope) this makes your `joint_effort_trajectory_controllers` obsolete. Maybe I am missing something, and you have a bit of different functionality of it which I didn't see scoping the code. Fell free to ask any questions about functionality or features on [ros2_controllers](https://github.com/ros-controls/ros2_controllers) repository.

P.S. Please don't hesitate to close this issue if you find it irrelevant.

ntorresalberto, 2022-06-09:
I'm not sure if it's related but I just tried to build this against ros2 and I'm getting a lot of ` is private within this context` from `joint_effort_trajectory_controller` files.

BarisYazici, 2023-07-31:
As soon as we have this https://github.com/ros-controls/ros2_controllers/pull/558 merged, we can migrate to the official JTC.

VladimirFokow, 2024-05-01:
Hi, I've noticed that this issue can proceed - https://github.com/ros-controls/ros2_controllers/pull/558 has been [merged](https://github.com/ros-controls/ros2_controllers/pull/558#event-10096178835),
and [backported](https://github.com/ros-controls/ros2_controllers/blob/humble/joint_trajectory_controller/CHANGELOG.rst#2300-2023-12-20) to `ros2_controllers` (humble) in 2.30.0 (2023-12-20).

TomCC7, 2024-06-03:
Hi, a stupid question here. How do I use the ros2_controller's `joint_trajectory_controller` package instead of the one in the repo? They share the same package name and even the controller name. Is there an elegant way to do that without modifying files in this repo?

VladimirFokow, 2024-06-03:
@TomCC7 I'm not advanced myself but here's my suggestion:

- delete (or add a file COLCON_IGNORE to) the `joint_trajectory_controller` folder of this repo

- install ros2-controllers, e.g.:
```bash
sudo apt update && sudo apt install ros-${ROS_DISTRO}-ros2-control ros-${ROS_DISTRO}-ros2-controllers 
```

- (remember to delete files that previous colcon build created (in build and install directories)), and  
  `rosdep install` and `colcon build`

TomCC7, 2024-06-03:
Thanks for replying! That certainly works and I'm currently using this method. Just don't want to configure this manually every time lol.

BarisYazici, 2025-01-31:
With the release 1.0.0, we removed our fork of JTC -> https://github.com/frankaemika/franka_ros2/releases/tag/v1.0.0.
User will need to use the ros-humble-joint-trajectory-controller

closed 2025-01-31 by BarisYazici