frankarobotics/franka_ros2 #8: Are there any plan to support ROS2 Humble?

CWEzio, 2022-08-15:
HI, I try to use franka_ros2 on Ubuntu 22.04, which only support ROS2 Humble. However, I encounter errors when I try to build this repo and it seems that currently ROS2 Humble is not supported. Are there any plan to support ROS2 Humble? Thank you! (I have to use Ubuntu 22.04 because my new computer hardware only supports Ubuntu 22.04, which is sad...)

andrejpan, 2022-08-18:
Until the end of the year, support for this will not happen.

mcbed, 2022-11-04:
Here is an as-is Humble port at PR #10

YuLiHN, 2022-11-29:
> Here is an as-is Humble port at PR #10

Thanks for your work! A small suggestion: the executable of control_manager in humble should be **spawn** rather than **spawn.py**
https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html

andrejpan, 2023-01-20:
In the next half a year it is highly likely there will be still no ROS2 support from FE.

peterdavidfagan, 2023-01-23:
> In the next half a year it is highly likely there will be still no ROS2 support from FE.

Thanks for the quick response on this @andrejpan and understood.

peterdavidfagan, 2023-01-26:
@mcbed thanks for your port to humble. Do you have details of the firmware and library versions you are using in order to run FER on ROS2?

andrejpan, 2023-02-05:
A degree of more transparency. The whole company goes to 90% short-term work in the incoming week for up to 5 months. ROS 2 support will be delayed accordingly.

peterdavidfagan, 2023-02-20:
@andrejpan @marcbone would it be possible to get guidance on what updates need to be performed to https://github.com/frankaemika/franka_ros2/pull/10 in order to run ROS2 Humble?

It would be very beneficial to run Humble on our system as we can then leverage latest changes to MoveIt (the new python library) in our research. I am willing to put some development time into this if you can provide guidance.

marcbone, 2023-02-21:
Honestly, I think nobody of us had the time to try out #10, which is a shame :( . But I have no reason to think that it does not work.

mcbed, 2023-02-21:
On my side, I do not have a Franka robot to test the code anymore but I will continue merging to PR #10 your suggestions to keep everything in one place. So do not hesitate to propose PRs to [this branch](https://github.com/mcbed/franka_ros2/tree/humble).

peterdavidfagan, 2023-02-21:
Thanks @marcbone and @mcbed, I have started testing on the Franka we have in our lab. I was encountering errors related to ROS2 control not finding the effort command interfaces. 

It's about time for me to dig deeper into ROS2 control anyways so I will look to explore this further. I think the changes @mcbed has made are likely to work with very minor tweaks.

peterdavidfagan, 2023-02-21:
Hi @cs-99, 

I am close to getting the latest JointTrajectoryController working on Humble. This controller is written to accept parameters from a yaml file the is compatible with [generate_parameter_library](https://github.com/PickNikRobotics/generate_parameter_library) ([see here](https://github.com/ros-controls/ros2_controllers/blob/5713a4149d4092eb3da30ad1d595ca15722c5294/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L59)).

I am only starting to work with the above myself and will be extremely cautious before running the controller on real hardware. Right now I plan to have a branch that I believe should be functional with the latest set of changes to ros2-control.

cs-99, 2023-02-28:
Hi @peterdavidfagan, did you manage to get it running?

peterdavidfagan, 2023-03-01:
Hi @cs-99, 

I did manage to get it running, the following pr contains much of the code: https://github.com/frankaemika/franka_ros2/pull/10.

andrejpan, 2023-04-19:
The company is finally giving resources for ROS 2 development of FR3. Unofficially, the limited release for Humble should arrive in ~3 months.

drewskoots, 2023-04-19:
Great to hear! Thanks for that update :) @peterdavidfagan Thanks for the suggestion, upon first glance it looks very promising and like it can tide me over until official support is released!

BarisYazici, 2023-07-31:
ROS 2 humble is supported now in the humble branch (default branch on github). Over time we will support more functionalities.

closed 2023-07-31 by BarisYazici