frankarobotics/franka_ros2 #11: [franka_descrpition] Inconsistent w/ ROS 1 variant - dynamics estimates?

EricCousineau-TRI, 2022-11-20:
Relates https://www.franka-community.de/t/panda-dynamic-model/4573/3

In `docs`, I see the following for ROS 1 flavor:
https://github.com/frankaemika/docs/blob/12fb63aa406b3e33dfc57a97ad5a5362e1ad8aad/source/franka_ros.rst#L36-L39

However, I don't see anything similar for ROS 2, at least upon examining this source tree and/or the docs
https://github.com/frankaemika/docs/blob/12fb63aa406b3e33dfc57a97ad5a5362e1ad8aad/source/franka_ros2.rst

Is there a reason why ROS 2 seems to lag / be different? If so, can it be documented?

Alternatively, is it possible to add this option?

marcbone, 2023-02-06:
As we currently do not support any simulator, we did not include the dynamics estimates. They will be included once we offer simulation support.

AndreasKuhner, 2024-10-30:
Hi @EricCousineau-TRI ,
nowadays we are supporting Gazebo - so, is this problem still persisting? If yes, don't hesitate to re-open the issue :smile:

closed 2024-10-30 by AndreasKuhner