frankarobotics/franka_ros2 #14: Compatible Library Versions

peterdavidfagan, 2023-01-23:
The following document only outlines version compatibility for franka_ros. 

I have updated my FER to 4.2.2 and wished to check what the recommended libfranka and associated submodule versions were for interfacing with the robot using franka_ros2.

Thanks in advance for any clarity you can provide. My workspace is built as a Docker image so I am happy to share further technical details where required.

marcbone, 2023-02-03:
libfranka 0.9.x will work for your robot. So installing 0.9.0 will work. I guess you still have a libfranka 0.8 somewhere on your system which shadows your 0.9 installation.

My random guess is that if you run:
```sudo apt remove ros-foxy-libfranka```
your problem will be solved. Feel free to reopen if this does not fix your issue.

In general all current libfranka versions should work with franka_ros2

closed 2023-02-03 by marcbone

peterdavidfagan, 2023-02-10:
Hi @marcbone, 

I retested today with 0.9.0 again and I am still facing the above error. I also am building libfranka from source, I also ran `sudo apt remove ros-foxy-libfranka` as suggested above. 

Here is a Dockerfile to recreate the issue:

```
ARG ROS_DISTRO=foxy
FROM ros:${ROS_DISTRO}-ros-base
MAINTAINER Peter David Fagan "peterdavidfagan@gmail.com"

# set default DDS to Cyclone
ENV RMW_IMPLEMENTATION rmw_cyclonedds_cpp

# install libfranka
RUN apt update && \
    apt install -y libpoco-dev libeigen3-dev && \
    git clone --branch 0.9.0 https://github.com/frankaemika/libfranka.git --recursive && \
    cd libfranka && \
    mkdir build && cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF  .. && \
    cmake --build . -j$(nproc) && \
    cpack -G DEB && \
    sudo dpkg -i libfranka-*.deb

# set up workspace
ENV ROS_UNDERLAY /root/panda_ws/install
WORKDIR $ROS_UNDERLAY/..
COPY . .

# import dependent libraries
RUN vcs import --skip-existing src < src/moveit2_tutorials/moveit2_tutorials.repos 

# install ros dependencies
RUN . /opt/ros/${ROS_DISTRO}/setup.sh && \
    apt-get -q update && \
    apt-get install -y ros-${ROS_DISTRO}-rmw-cyclonedds-cpp && \
    apt-get install -y \
    ros-foxy-control-msgs \
    ros-foxy-xacro \
    ros-foxy-angles \
    ros-foxy-ros2-control \
    ros-foxy-realtime-tools \
    ros-foxy-control-toolbox \
    ros-foxy-moveit \
    ros-foxy-ros2-controllers \
    ros-foxy-joint-state-publisher \
    ros-foxy-joint-state-publisher-gui \
    ros-foxy-ament-cmake-clang-format && \
    rosdep update && \
    DEBIAN_FRONTEND=noninteractive \
    rosdep install --from-paths src --ignore-src -r -y --rosdistro ${ROS_DISTRO} --as-root=apt:false && \ 
    colcon build --mixin debug && \
    # Clear apt-cache to reduce image size
    rm -rf /var/lib/apt/lists/* && \
    # Update /ros_entrypoint.sh to source our new workspace
    sed -i "s#/opt/ros/\$ROS_DISTRO/setup.bash#$ROS_UNDERLAY/setup.sh#g" /ros_entrypoint.sh
```

marcbone, 2023-02-13:
I can assure you that you still have 0.8.0 somewhere on your system. Otherwise the error would not show that you have library version 4. You can verify this by putting
```cmake
message("libfranka version: ${Franka_VERSION}")
```
into your CMakeLists.txt.

peterdavidfagan, 2023-02-13:
Thanks for getting back to me on this @marcbone. Ok I'll run the container again and look for a shared library file for 0.8.0.

From the above Dockerfile it seems surprising that this version of libfranka is getting installed on the system.

peterdavidfagan, 2023-02-13:
The following builds and runs as expected. As @marcbone outlined I was still installing 0.8.0 via rosdep.

```
ARG ROS_DISTRO=foxy
FROM ros:${ROS_DISTRO}-ros-base
MAINTAINER Peter David Fagan "peterdavidfagan@gmail.com"

# set default DDS to Cyclone
ENV RMW_IMPLEMENTATION rmw_cyclonedds_cpp

# install libfranka
RUN apt update && apt upgrade -y && \
    apt install -y  build-essential cmake git libpoco-dev libeigen3-dev && \
    git clone --recursive https://github.com/frankaemika/libfranka.git --branch 0.9.0 && \
    cd libfranka && \
    mkdir build && cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF  .. && \
    cmake --build . -j$(nproc) && \
    cpack -G DEB && \
    sudo dpkg -i libfranka-*.deb

# set up workspace
ENV ROS_UNDERLAY /root/panda_ws/install
WORKDIR $ROS_UNDERLAY/..
COPY . .

# import dependent libraries
RUN vcs import --skip-existing src < src/moveit2_tutorials/moveit2_tutorials.repos 

# install ros dependencies
RUN . /opt/ros/${ROS_DISTRO}/setup.sh && \
    apt-get -q update && \
    apt-get install -y ros-${ROS_DISTRO}-rmw-cyclonedds-cpp && \
    apt-get install -y \
    ros-foxy-control-msgs \
    ros-foxy-xacro \
    ros-foxy-angles \
    ros-foxy-ros2-control \
    ros-foxy-realtime-tools \
    ros-foxy-control-toolbox \
    ros-foxy-moveit \
    ros-foxy-ros2-controllers \
    ros-foxy-joint-state-publisher \
    ros-foxy-joint-state-publisher-gui \
    ros-foxy-ament-cmake-clang-format && \
    rosdep update && \
    DEBIAN_FRONTEND=noninteractive \
    rosdep install --from-paths src --ignore-src -r -y --skip-keys "libfranka" --rosdistro ${ROS_DISTRO} && \ 
    colcon build --mixin debug && \
    # Clear apt-cache to reduce image size
    rm -rf /var/lib/apt/lists/* && \
    # Update /ros_entrypoint.sh to source our new workspace
    sed -i "s#/opt/ros/\$ROS_DISTRO/setup.bash#$ROS_UNDERLAY/setup.sh#g" /ros_entrypoint.sh
```