frankarobotics/franka_ros2 #15: franka moveit crushes after the external activation device is pressed down

YuLiHN, 2023-02-03:
While the terminal is running 
`ros2 launch franka_moveit_config moveit.launch.py robot_ip:=my_robot_ip`
and I press external activation device which makes the display light turn from blue to white, the moveit node crushes because of the exception `what(): libfranka: Move command aborted: User Stop pressed!`. But in ros1 it is not the case. In ros1 the moveit node is still running after I bring the external activation device to the activated state (white light).

my terminal is like this:

[INFO] [launch]: All log files can be found below /home/rpl/.ros/log/2023-02-03-11-10-03-834375-robotics-261820
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [261824]
[INFO] [robot_state_publisher-2]: process started with pid [261826]
[INFO] [move_group-3]: process started with pid [261828]
[INFO] [ros2_control_node-4]: process started with pid [261830]
[INFO] [joint_state_publisher-5]: process started with pid [261832]
[INFO] [franka_gripper_node-6]: process started with pid [261834]
[INFO] [ros2 run controller_manager spawner.py panda_arm_controller-7]: process started with pid [261836]
[INFO] [ros2 run controller_manager spawner.py joint_state_broadcaster-8]: process started with pid [261838]
[robot_state_publisher-2] Parsing robot urdf xml string.
[franka_gripper_node-6] [INFO] [1675419004.204783183] [panda_gripper]: Trying to establish a connection with the gripper
[franka_gripper_node-6] [INFO] [1675419004.205655200] [panda_gripper]: Connected to gripper
[robot_state_publisher-2] Link panda_link0 had 1 children
[robot_state_publisher-2] Link panda_link1 had 1 children
[robot_state_publisher-2] Link panda_link2 had 1 children
[robot_state_publisher-2] Link panda_link3 had 1 children
[robot_state_publisher-2] Link panda_link4 had 1 children
[robot_state_publisher-2] Link panda_link5 had 1 children
[robot_state_publisher-2] Link panda_link6 had 1 children
[robot_state_publisher-2] Link panda_link7 had 1 children
[robot_state_publisher-2] Link panda_link8 had 1 children
[robot_state_publisher-2] Link panda_hand had 3 children
[robot_state_publisher-2] Link panda_leftfinger had 0 children
[robot_state_publisher-2] Link panda_rightfinger had 0 children
[robot_state_publisher-2] Link panda_hand_tcp had 0 children
[robot_state_publisher-2] [INFO] [1675419004.206866968] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-2] [INFO] [1675419004.206939973] [robot_state_publisher]: got segment panda_hand_tcp
[robot_state_publisher-2] [INFO] [1675419004.206952427] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-2] [INFO] [1675419004.206962090] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-2] [INFO] [1675419004.206971538] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-2] [INFO] [1675419004.206980825] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-2] [INFO] [1675419004.206990021] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-2] [INFO] [1675419004.206999166] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-2] [INFO] [1675419004.207008518] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-2] [INFO] [1675419004.207017768] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-2] [INFO] [1675419004.207026967] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-2] [INFO] [1675419004.207036385] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-2] [INFO] [1675419004.207045645] [robot_state_publisher]: got segment panda_rightfinger
[robot_state_publisher-2] [INFO] [1675419004.207055537] [robot_state_publisher]: got segment panda_table_link
[ros2_control_node-4] [INFO] [1675419004.219853161] [FrankaHardwareInterface]: Connecting to robot at "172.16.1.22" ...
[ros2_control_node-4] [INFO] [1675419004.221130550] [FrankaHardwareInterface]: Successfully connected to robot
[ros2_control_node-4] [INFO] [1675419004.221264865] [FrankaHardwareInterface]: Started
[ros2_control_node-4] [INFO] [1675419004.231040806] [controller_manager]: update rate is 1000 Hz
[move_group-3] [WARN] [1675419004.259931590] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-3] [WARN] [1675419004.259978652] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-3] Parsing robot urdf xml string.
[move_group-3] [INFO] [1675419004.264849176] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00253798 seconds
[move_group-3] [INFO] [1675419004.264884871] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[move_group-3] [WARN] [1675419004.264991459] [moveit_robot_model.robot_model]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[move_group-3] [WARN] [1675419004.265001741] [moveit_robot_model.robot_model]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[move_group-3] Link panda_link0 had 1 children
[move_group-3] Link panda_link1 had 1 children
[move_group-3] Link panda_link2 had 1 children
[move_group-3] Link panda_link3 had 1 children
[move_group-3] Link panda_link4 had 1 children
[move_group-3] Link panda_link5 had 1 children
[move_group-3] Link panda_link6 had 1 children
[move_group-3] Link panda_link7 had 1 children
[move_group-3] Link panda_link8 had 1 children
[move_group-3] Link panda_hand had 3 children
[move_group-3] Link panda_leftfinger had 0 children
[move_group-3] Link panda_rightfinger had 0 children
[move_group-3] Link panda_hand_tcp had 0 children
[... 113 lines cut ...]
[ros2_control_node-4]   what():  libfranka: Move command aborted: User Stop pressed!
[ERROR] [ros2_control_node-4]: process has died [pid 261830, exit code -6, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_u3tunc1p --params-file /home/rpl/franka_ros2_ws/install/franka_moveit_config/share/franka_moveit_config/config/panda_ros_controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-4] was required: shutting down launched system
[INFO] [franka_gripper_node-6]: sending signal 'SIGINT' to process[franka_gripper_node-6]
[INFO] [joint_state_publisher-5]: sending signal 'SIGINT' to process[joint_state_publisher-5]
[INFO] [move_group-3]: sending signal 'SIGINT' to process[move_group-3]
[INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
[INFO] [rviz2-1]: sending signal 'SIGINT' to process[rviz2-1]
[franka_gripper_node-6] [INFO] [1675419011.078615591] [rclcpp]: signal_handler(signal_value=2)
[move_group-3] [INFO] [1675419011.080601808] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-2] [INFO] [1675419011.082003243] [rclcpp]: signal_handler(signal_value=2)
[rviz2-1] [INFO] [1675419011.083257282] [rclcpp]: signal_handler(signal_value=2)
[move_group-3] [INFO] [1675419011.090332322] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-3] [INFO] [1675419011.091618592] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-3] [INFO] [1675419011.092063162] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[move_group-3] [INFO] [1675419011.092272626] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 261826]
[move_group-3] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[move_group-3]          at line 127 in /tmp/binarydeb/ros-foxy-class-loader-2.0.3/src/class_loader.cpp
[INFO] [move_group-3]: process has finished cleanly [pid 261828]
[move_group-3] 
[INFO] [joint_state_publisher-5]: process has finished cleanly [pid 261832]
[rviz2-1] [INFO] [1675419011.132836383] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[INFO] [franka_gripper_node-6]: process has finished cleanly [pid 261834]
[INFO] [rviz2-1]: process has finished cleanly [pid 261824]

YuLiHN, 2023-02-03:
> This is expected behavior. Maybe you can elaborate a bit, what are you trying to achieve, so we can think about changing this in the future. I guess you want to be able to guide the robot and only want to control the robot when you are executing a trajectory in MoveIt2. Is that right?

Thanks for the fast answer.
And yes, when we are planing&executing some trajectories in Moveit2, there could be some failure cases. We want to guide the robot back to the start position manually by bringing to robot to activated state, and then continue to use Moveit2 node. Otherwise it takes a lot of time to restart the Moveit2.

marcbone, 2023-02-03:
So the problem is, that when you launch the MoveIt, it will start a controller that keeps the robot in its current position. When you then press the enabling device, the controller dies. You can try to disable the controller using  the ros2 control cli, then guide the robot and then enable the controller again. But I am not sure if that works. What would definitely work is to move the robot back using MoveIt

YuLiHN, 2023-02-04:
> So the problem is, that when you launch the MoveIt, it will start a controller that keeps the robot in its current position. When you then press the enabling device, the controller dies. You can try [...]

Thanks for your advice, we will add a function that moves the robot back using Moveit2. Closing the issue

closed 2023-02-04 by YuLiHN

marcbone, 2023-02-06:
There is also the predefined ready pose that you can use directly as goal pose in Rviz