frankarobotics/franka_ros2 #18: Franka Description Package

peterdavidfagan, 2023-04-21:
I noticed that there is a discrepancy between the `franka_description` package for the ROS and ROS 2 repositories. 

Are there plans to port changes? Is it worth opening a pr?

BarisYazici, 2023-04-21:
@peterdavidfagan  can you be more specific? For example, what are the features you need. So we can take a better look in the planning on Monday. If you think it's worthy of a PR, why not :)

peterdavidfagan, 2023-04-21:
Thanks for the quick response on this @BarisYazici. 

I am currently working on the `rolling-devel` branch for [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config). One of the first items I came across as missing was links from the URDF. Specifically the macro defined [here](https://github.com/frankaemika/franka_ros/blob/cf539562038e3404ac48f76b70b13be9dfa8c504/franka_description/robots/common/utils.xacro#L38) and its usage in this [xacro file](https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/common/franka_arm.xacro) help ensure compatibility with MoveIt SRDF files in [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config).

peterdavidfagan, 2023-04-21:
Ok I have started some work on this in the following https://github.com/frankaemika/franka_ros2/pull/19.

BarisYazici, 2023-04-25:
Hey @peterdavidfagan, everything looks really good! Thanks a lot for investing your time to make it work! We had our planning meeting yesterday, our priorities will be:

1) Write tests for the existing functionality.
2) Verify #10 on FR3 and merge
3) Implement ReadBlockOnce function in libfranka and integrate it to this repo
4) And then we will merge #19 PR. 

This plan is for the next 3 weeks. In the meantime, we would appreciate if someone from other teams at Franka or from contributors to give your PR a review. If we find time, we will for sure give you some feedback.

peterdavidfagan, 2023-04-25:
Thanks @BarisYazici, excited for the above development items.

closed 2025-03-05 by AndreasKuhner