frankarobotics/franka_ros2 #20: collision meshes not shown in RViz

christian-rauch, 2023-04-27:
The collision geometry defined in the XACRO is not visualised in RViz.

# Steps to reproduce:
1. follow instructions at https://support.franka.de/docs/franka_ros2.html
2. start the fake robot: `ros2 launch franka_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true`
3. show collision geometry: tick `PlanningScene` -> `Scene Robot` -> `Show Robot Collision` and untick `Show Robot Visual`

# Expected Results:

You should see the collision geometry, such as `<cylinder radius="${0.06+safety_distance}" length="0.03" />` for link `"${arm_id}_link0"` in RViz.

# Actual Results:

No collision geometry is shown.

MoveIt seems to respect the collision geometry as it prevents collision states, but they are not shown in RViz.

christian-rauch, 2023-04-27:
I overlooked the `Optional .bashrc Settings` part.

The workaround for this issue is to reset `LC_NUMERIC`, i.e.:
```sh
LC_NUMERIC= ros2 launch franka_moveit_config moveit.launch.py robot_ip:=dont-care
```

BarisYazici, 2023-08-01:
Can you visualize the collision geometries on rviz now?

christian-rauch, 2023-08-01:
> Can you visualize the collision geometries on rviz now?

Yes, but only with the workaround above. This is not a proper fix and I would like to keep this open until fixed.

BarisYazici, 2023-08-01:
Can this be cause https://github.com/ros-planning/moveit2_tutorials/issues/508?
In my setup, I can visualize the collisions. Can you maybe check with the .devcontainer or the dockerfile we provide in the `humble`  branch

christian-rauch, 2023-08-01:
> I think the issue is fixed [here](https://github.com/ros/urdfdom_headers/pull/42), so I guess the next version of ROS 2 will not have the problem. However, since its definitely not something that we can fix in our repo, I would like to close this issue

This was merged more than 5 years ago and I am pretty sure that this fix already made it into released versions.

christian-rauch, 2023-08-01:
> Is this the cause [ros-planning/moveit2_tutorials#508](https://github.com/ros-planning/moveit2_tutorials/issues/508)?

This problem and the one discussed here may have the same cause. Most likely, both issues are caused by the different decimal point conventions in different locales.

> In my setup, I can visualize the collisions.

What does your setup look like? Are you running "native" via the ROS packages outside of a Docker container? Which locale are you using?

> Can you maybe check with the .devcontainer or the dockerfile we provide in the `humble` branch?

The Docker image seems to only build libfranka and does nothing with respect to the ROS node. In any case, it may work inside Docker with a different locale, but most people will use it as part of their colcon workspace so it has to work with any locale.

BarisYazici, 2023-08-02:
I suspect this might be an issue in parameter parser as mentioned in the  https://github.com/ros-planning/moveit2_tutorials/issues/508.

Could you try this?
```
ros2 launch franka_description visualize_franka.launch.py
```

And go under `robot_model` and tick the collision_enabled checkbox. Can you see the meshes then?

AndreasKuhner, 2024-10-30:
Hi @christian-rauch ,
is this issue still persisting? If yes, don't hesitate to re-open the issue :smile:

closed 2024-10-30 by AndreasKuhner