frankarobotics/franka_ros2 #21: ROS2 Humble compile error

prasuchit, 2023-05-02:
Hi,

I followed the instructions on [this page](https://frankaemika.github.io/docs/franka_ros2.html) to install everything correctly on ROS2 Humble. However, when I do:

     sudo rosdep install --from-paths . --ignore-src -r -y --rosdistro=humble --skip-keys libfranka

I get the error: Unable to locate package ros-humble-warehouse-ros-mongo

I couldn't find any help online for this or any alternative way to install this package. Kindly help. Any guidance would be appreciated.

cs-99, 2023-05-02:
This repository works with ROS 2 Foxy, but it seems like there will be support for Humble in the near future (see #8). For now, there is a humble port at #10.

prasuchit, 2023-05-03:
@cs-99 Thanks for the prompt reply, I just tried the pr #10 version and it still throws the same error:

     E: Unable to locate package ros-humble-warehouse-ros-mongo
     
I think this is a Moveit2 issue that they haven't ported the mongo warehouse to Humble. I don't know what to do at this point. :(

cs-99, 2023-05-03:
Do you want to use the warehouse? If not, you should be able to just ignore this error (and remove the package from launch files and dependencies if you want to).

prasuchit, 2023-05-03:
@cs-99 That's what I ended up doing just now. I thought it was essential for the package to work, but I found that the moveit.launch file works fine without it. Thanks for the help. :)

BarisYazici, 2023-07-28:
Humble support is ready. It is the main brach now. @prasuchit

closed 2023-07-28 by BarisYazici