frankarobotics/franka_ros2 #22: Incompatible library version

kuyhnooj, 2023-05-10:
Sorry for poor English ;(

I'm going to control Franka Research 3 with ROS2(foxy).

My franka world system version is 5.3.0 and I installed libfranka fr3-develop.

However i wrote down ros2 launch franka_moveit_config moveit_launch.py robot_ip:=my_ip,
i received this message,  
[FATAL] [1683694173.610288313] [FrankaHardwareInterface]: Could not connect to robot
[ros2_control_node-4] [FATAL] [1683694173.610295895] [FrankaHardwareInterface]: libfranka: Incompatible library version (server version: 6, library version: 5). Please check https://frankaemika.github.io for Panda system updates or choose a libfranka version that uses the server version 6 from the table at https://frankaemika.github.io/docs/compatibility.html .

I checked libfranka/common/include/research_interface/robot/service_types.h, constexpr Version kVersion = 6;.

I have no idea to make things right. 

Additionally, this ROS2 source is only for panda. So i edited all xacro and file and name 'panda' to 'fr3'.

Do you have any plan to finish ROS2 source for FR3??

andrejpan, 2023-05-10:
Redundant issue #8

closed 2023-05-10 by andrejpan

marcbone, 2023-05-10:
You can edit replace panda with fr3 everywhere but this does not do much. The current version works with panda and fr3 on foxy