frankarobotics/franka_ros2 #23: controller manager crashes when controller is stopped

christian-rauch, 2023-05-10:
After loading the `joint_state_broadcaster` and `panda_arm_controller` via `ros2 run controller_manager spawner.py` into the controller manager, I am unable to unload `panda_arm_controller` again.

An attempt to unload via:
```sh
ros2 run controller_manager unspawner.py panda_arm_controller
```
or simply stop via:
```sh
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController "{stop_controllers: ['panda_arm_controller'], strictness: 1}"
```
crashes the controller manager:
```
[ERROR] [ros2_control_node-6]: process has died [pid 3810693, exit code -11, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_a2hvyo1e --params-file [...]/install/franka_moveit_config/share/franka_moveit_config/config/panda_ros_controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-6] was required: shutting down launched system
```

BarisYazici, 2023-08-31:
Can you tell me how you spawn the controller?

can you also try this with the ros2_control cli commands?

so for example

first load the controller

`ros2 control load_controller joint_impedance_example_controller`

`ros2 control unload_controller joint_impedance_example_controller`

christian-rauch, 2023-09-05:
I am launching the node via `ros2 launch franka_moveit_config moveit.launch.py robot_ip:=172.16.0.2`. Looking at the loaded controllers via `ros2 control list_controllers` I can see that only the `joint_state_broadcaster` and `panda_arm_controller` are loaded by default:
```
panda_arm_controller[joint_effort_trajectory_controller/JointTrajectoryController] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
```

An additional `ros2 control load_controller joint_impedance_example_controller` will give me `[controller_manager]: The 'type' param not defined for 'joint_impedance_example_controller'.`.

Unloading the `panda_arm_controller` via `ros2 control unload_controller panda_arm_controller` does not work as it is running:
```
[controller_manager]: Could not unload controller with name 'panda_arm_controller' because it is still running
```
and there is no direct CLI command to stop the controller (`ros2 control -h` does not list an command to stop the controller only to start via `configure_start_controller`).

In any case, doing a `ros2 run controller_manager unspawner.py panda_arm_controller` still gives me:
```
[ERROR] [ros2_control_node-4]: process has died [pid 764118, exit code -11, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_8_crup7w --params-file [workspace]/install/franka_moveit_config/share/franka_moveit_config/config/panda_ros_controllers.yaml -r joint_states:=franka/joint_states'].
```

BarisYazici, 2023-10-12:
set the controller state to inactive first before unloading. I think this issue is more related to the controller_manager.

`ros2 control set_controller_state panda_arm_controller inactive`

then 

`ros2 control unload_controller panda_arm_controller `

closed 2023-10-12 by BarisYazici

christian-rauch, 2023-10-17:
> `ros2 control set_controller_state panda_arm_controller inactive`

Is `inactive` a `humble` feature? I tried this before on `foxy` and got:
```
$ ros2 control set_controller_state panda_arm_controller inactive
usage: ros2 control set_controller_state [-h] [--spin-time SPIN_TIME]
                                         [-c CONTROLLER_MANAGER]
                                         [--include-hidden-nodes]
                                         controller_name
                                         {configure,start,stop}
ros2 control set_controller_state: error: argument state: invalid choice: 'inactive' (choose from 'configure', 'start', 'stop')
```