frankarobotics/franka_ros2 #28: Franka_example_controllers Segmentation Faults upon colcon build

drewskoots, 2023-08-01:
Hello,

Ive had it a few times now that my colcon build task fails with a segmentation fault upon building franka_example_controllers. I'm not quite sure if this is the correct point in stack to log this error, if not, please let me know where would be more suitable.

Relevant `colcon build` output:
```
--- stderr: franka_example_controllers         
In file included from /usr/include/eigen3/Eigen/Core:281,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /fog_ws/src/miti-ros/src/franka_ros2/franka_example_controllers/include/franka_example_controllers/joint_impedance_example_controller.hpp:19,
                 from /fog_ws/src/miti-ros/src/franka_ros2/franka_example_controllers/src/joint_impedance_example_controller.cpp:15:
/usr/include/c++/9/functional: In instantiation of 'std::_Bind<_Functor(_Bound_args ...)>::_Bind(std::_Bind<_Functor(_Bound_args ...)>&&) [with _Functor = std::function<void(long unsigned int)>; _Bound_args = {std::_Placeholder<1>}]':
/usr/include/c++/9/bits/std_function.h:252:39:   required from 'static void std::_Function_base::_Base_manager<_Functor>::_M_init_functor(std::_Any_data&, _Functor&&, std::false_type) [with _Functor = std::_Bind<std::function<void(long unsigned int)>(std::_Placeholder<1>)>; std::false_type = std::integral_constant<bool, false>]'
/usr/include/c++/9/bits/std_function.h:223:19:   required from 'static void std::_Function_base::_Base_manager<_Functor>::_M_init_functor(std::_Any_data&, _Functor&&) [with _Functor = std::_Bind<std::function<void(long unsigned int)>(std::_Placeholder<1>)>]'
/usr/include/c++/9/bits/std_function.h:675:34:   required from 'std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = std::_Bind<std::function<void(long unsigned int)>(std::_Placeholder<1>)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = void; _ArgTypes = {long unsigned int, int}]'
/opt/ros/humble/rclcpp/include/rclcpp/rclcpp/publisher_base.hpp:298:94:   required from here
/usr/include/c++/9/functional:474:9: internal compiler error: Segmentation fault
  474 |       { }
      |         ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/franka_example_controllers.dir/build.make:90: CMakeFiles/franka_example_controllers.dir/src/joint_impedance_example_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/franka_example_controllers.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< franka_example_controllers [1min 7s, exited with code 2]
```

BarisYazici, 2023-08-04:
I am not sure what the problem is. There might be a version mismatch with the eigen ros uses and eigen you have locally. But just guessing. You can try to install this package eigen3-cmake-module `sudo apt install ros-humble-eigen3-cmake-module`

drewskoots, 2023-08-07:
I'm running Humble on Ubuntu 20.04 so I have to compile everything from source, so the version mismatch sounds plausible. I've also noticed that RAM usage is really high at this point in the build, so maybe the swap just runs out and it crashes. It does seem unlikely that franka_ros is the cause, so I'll close the issue.

closed 2023-08-07 by drewskoots