frankarobotics/franka_ros2 #29: Colon build failed even when Prerequisites were done

Hydrogenplus3, 2023-08-04:
Hello! I would like to report the issue that  I met when running the Clone repo and build packages process. I guess the error may happen due to the missing of backward_ros during ros2 installation.

[image]

The error meassage is shown as following:

`:~/franka_ros2_ws$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[1.669s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'joint_trajectory_controller' is in: /opt/ros/foxy
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may resu in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding joint_trajectory_controller

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> franka_msgs
Starting >>> franka_description
Starting >>> integration_launch_testing
Starting >>> joint_trajectory_controller                                                                               
Finished <<< integration_launch_testing [0.71s]                                                          
Finished <<< franka_description [0.79s]                                                                  
--- stderr: joint_trajectory_controller                                                                      
CMake Error at CMakeLists.txt:29 (find_package):
  By not providing "Findbackward_ros.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "backward_ros", but CMake did not find one.

  Could not find a package configuration file provided by "backward_ros" with
  any of the following names:

    backward_rosConfig.cmake
    backward_ros-config.cmake

  Add the installation prefix of "backward_ros" to CMAKE_PREFIX_PATH or set
  "backward_ros_DIR" to a directory containing one of the above files.  If
  "backward_ros" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< joint_trajectory_controller [1.18s, exited with code 1]
Aborted  <<< franka_msgs [2.11s]                                 
                                 
Summary: 2 packages finished [2.93s]
  1 package failed: joint_trajectory_controller
  1 package aborted: franka_msgs
  1 package had stderr output: joint_trajectory_controller
  7 packages not processed
`

BarisYazici, 2023-08-04:
can you install `sudo apt install ros-humble-backward-ros` and try again?

BarisYazici, 2023-08-04:
Also it seems like your system still has `ros2 foxy` you need to install `ros2 humble` to use the default branch.
If you have vscode you can try building the repo in a devcontainer. The settings for the devcontainer we provided in the humble branch.

huangwc94, 2023-08-04:
I am still getting error:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
franka_gripper: Cannot locate rosdep definition for [libfranka]
```
when using dev container provided by yours

BarisYazici, 2023-08-04:
> I am still getting error:
> 
> ```
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> franka_gripper: Cannot locate rosdep definition for [libfranka]
> ```
> 
> when using dev container provided by yours. Seems like there is no need to do `rosdep install --from-path src` in this case

You don’t need to do rosdep install you can just build it by `colcon build` and source then start the controller

huangwc94, 2023-08-04:
Seems like https://github.com/frankaemika/franka_ros2/blob/humble/franka_gripper/package.xml#L21C29-L21C29 can be comment out for now as there is no libfranka ros package available

Hydrogenplus3, 2023-08-05:
> Thank you for your quick respond! However, I’m unsure if I understand your intention correctly. I followed the instructions on the official website https://frankaemika.github.io/docs/franka_ros2.html, which stated that the supported version is Foxy for Ubuntu 20.04. Are you referring to ROS Humble as a version specifically for Ubuntu 22.04? Or is it a package that is compatible with Foxy? I would appreciate some clarification on this matter.

Hydrogenplus3, 2023-08-05:
> Also it seems like your system still has `ros2 foxy` you need to install `ros2 humble` to use the default branch. If you have vscode you can try building the repo in a devcontainer. The settings for the devcontainer we provided in the humble branch.

Thank you for your quick response! However, I’m unsure if I understand your intention correctly. I followed the instructions on the official website https://frankaemika.github.io/docs/franka_ros2.html, which stated that the supported version is Foxy for Ubuntu 20.04. Are you referring to ROS Humble as a version specifically for Ubuntu 22.04? Or is it a package that is compatible with Foxy? I am rather confused and would appreciate some clarification on this matter.

BarisYazici, 2023-08-05:
sure I can clarify, we recently released the humble version and haven’t yet had time to update the documentation. This documentation is still for the old foxy version. You are right humblel requires ubuntu 22.04. 
ROS 2 has different distros and Foxy is the distro before Humble. I hope this clarifies.
Closing because problem seems to be resolved

closed 2023-08-05 by BarisYazici