frankarobotics/franka_ros2 #31: franka_ros2 compatibility

saivojjala, 2023-08-23:
Hello, I wanted to know if the ros2 humble branch is up to date and can be used. The documentation states that franks ros 2 is a beta software and the installation is provided for ros2 foxy.

drewskoots, 2023-08-24:
Yes, the humble port is usable. It's a fairly recent release so I think the Documentation just hasn't been updated yet.

saivojjala, 2023-08-25:
Hi @drewskoots , the humble port throws the following error while building franka_hardware package:

Could not find a configuration file for package "Franka" that is compatible
  with requested version "0.11.0".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/Franka/FrankaConfig.cmake, version: 0.9.2
    /lib/cmake/Franka/FrankaConfig.cmake, version: 0.9.2
    
However, when I attempt to install libfranka 0.11.0, the branch is not found. I changed the version in the CMake to 0.10.0 and 0.9.2, but then the following error occurs:

fatal error: franka/active_control.h: No such file or directory
   25 | #include <franka/active_control.h>

Could you please take a look into this issue.

BarisYazici, 2023-08-27:
you need to switch to the fr3-develop branch in the libfranka. Latest release of franka-ros2 depends on fr3-develop branch. https://github.com/frankaemika/franka_ros2/releases/tag/v0.1.3

saivojjala, 2023-08-27:
okay thanks! it worked now

closed 2023-08-27 by saivojjala