frankarobotics/franka_ros2 #33: joint_trajectory_controller throws ["max_path_pose_deviation_violation"] error

tkelestemur-bdai, 2023-09-07:
I'm using the `joint_trajectory_follower` with its action server. I send a single waypoint to the trajectory controller and pretty often I get the following error and it kills the controller node. I've checked libfranka website but couldn't find anything. Below, you can see how I contrsuct the action goal and stack traceback of the error. I'm on the humble branch and using Ubuntu 22.04 with RT kernel `5.15.0-1045-realtime`. 

```python
    def send_trajectory(self, q: Union[Sequence[float], npt.NDArray], time_from_start: float = 2.5) -> None:
        goal_msg = FollowJointTrajectory.Goal()
        goal_msg.trajectory.joint_names = self.joint_names
        traj_point = JointTrajectoryPoint()
        traj_point.positions = []
        for qi in q:
            traj_point.positions.append(qi)
        traj_point.velocities = [0.0] * 7
        traj_point.accelerations = [0.0] * 7
        traj_point.time_from_start = rclpy.duration.Duration(seconds=time_from_start).to_msg()
        goal_msg.trajectory.points.append(traj_point)
        self.arm_trajectory_cli.send_goal(goal_msg)
```

```bash
[ros2_control_node-3]   what():  libfranka: Move command aborted: motion aborted by reflex! ["max_path_pose_deviation_violation"]
[ros2_control_node-3] Stack trace (most recent call last) in thread 444761:
[ros2_control_node-3] #18   Object "", at 0xffffffffffffffff, in
[ros2_control_node-3] #17   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f82f04179ff]
[ros2_control_node-3] #16   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f82f0385b42]
[ros2_control_node-3] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f82f0615252, in
[ros2_control_node-3] #14   Source "./src/ros2_control_node.cpp", line 80, in operator() [0x556281439800]
[ros2_control_node-3] #13   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f82f0221a81, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f82f0242994, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #11   Object "/home/tkelestemur/franka_ws/install/franka_hardware/lib/libfranka_hardware.so", at 0x7f82e9105498, in franka_hardware::FrankaHardwareInterface::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #10   Object "/home/tkelestemur/franka_ws/install/franka_hardware/lib/libfranka_hardware.so", at 0x7f82e9116278, in franka_hardware::Robot::readOnce()
[ros2_control_node-3] #9    Object "/home/tkelestemur/franka_ws/install/franka_hardware/lib/libfranka_hardware.so", at 0x7f82e9114f8e, in franka_hardware::Robot::readOnceActiveControl()
[ros2_control_node-3] #8    Object "/usr/lib/libfranka.so.0.11.0", at 0x7f82e904433d, in franka::ActiveControl::readOnce()
[ros2_control_node-3] #7    Object "/usr/lib/libfranka.so.0.11.0", at 0x7f82e90429c3, in franka::Robot::Impl::throwOnMotionError(franka::RobotState const&, unsigned int) [clone .cold]
[ros2_control_node-3] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f82f05e74d7, in __cxa_throw
[ros2_control_node-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f82f05e7276, in std::terminate()
[ros2_control_node-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f82f05e720b, in
[ros2_control_node-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f82f05dbb9d, in
[ros2_control_node-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f82f03197f2]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f82f0333475]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f82f0387a7c]
[ros2_control_node-3] Aborted (Signal sent by tkill() 444628 770001229)
[ERROR] [ros2_control_node-3]: process has died [pid 444628, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_i_dofwnd --params-file /home/tkelestemur/franka_ws/install/franka_bringup/share/franka_bringup/config/controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-3] was required: shutting down launched system
[INFO] [franka_gripper_node-6]: sending signal 'SIGINT' to process[franka_gripper_node-6]
[INFO] [joint_state_publisher-2]: sending signal 'SIGINT' to process[joint_state_publisher-2]
[INFO] [robot_state_publisher-1]: sending signal 'SIGINT' to process[robot_state_publisher-1]
[franka_gripper_node-6] [INFO] [1694108198.096681998] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1694108198.097610885] [rclcpp]: signal_handler(signum=2)
[INFO] [joint_state_publisher-2]: process has finished cleanly [pid 444626]
[INFO] [franka_gripper_node-6]: process has finished cleanly [pid 444634]
[ERROR] [robot_state_publisher-1]: process has died [pid 444624, exit code -11, cmd '/opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher --params-file /tmp/launch_params_3qs92vck'].

BarisYazici, 2023-10-25:
which command interface are you using?

tkelestemur-bdai, 2023-10-31:
We are running the joint_trajectory_controller as it is so it's using the [effort command interface](https://github.com/frankaemika/franka_ros2/blob/humble/franka_bringup/config/controllers.yaml#L47) Are there other command interfaces implemented? I thought it only works with effort interface.

BarisYazici, 2023-10-31:
Do you experience the same issue when you use the joint trajectory controller with MoveIt launch file in our repo?

tkelestemur-bdai, 2023-11-01:
We didn't test it a lot with MoveIt but with limited experience, we didn't see that error. We double checked our trajectory optimization method and it does not produce results that are out of the limits of FR3.

BarisYazici, 2023-11-06:
we will look into that issue. I will let you know the process.

BarisYazici, 2023-11-09:
I haven't tested yet but I have a suspicion that the [kDefaultDeviation](https://github.com/frankaemika/libfranka/blob/d0c9f3ce102ad6f26fbbc767a9895d85077ba668/src/control_loop.h#L20) value used in the libfranka might be causing this issue. Although you are using the torque control, we are passing the joint velocity default deviation to the [startMotion()](https://github.com/frankaemika/libfranka/blob/d0c9f3ce102ad6f26fbbc767a9895d85077ba668/src/robot.cpp#L277C23-L277C23). I think you can try increasing that the kDefaultDeviation and see if it's solving your issue, until we find an official solution.

BarisYazici, 2023-12-01:
We found the issue. It seems to be related to the pose deviation monitoring under the hood. So when the `O_T_EE(measured_pose)` and `O_T_EE_d(desired_pose)` deviates we trigger the error. In external control mode, we are sending [zero velocities to the robot](https://github.com/frankaemika/libfranka/blob/f3b8d775a9c847cab32684c8a316f67867761674/src/robot.cpp#L51), which in turn does not update the `O_T_EE_d` and triggers to error.
 
The real fix is disabling the pose deviation monitoring in the `External Control Mode`, and we are working on that. But as a workaround we provide [0.13.1](https://github.com/frankaemika/libfranka/releases/tag/0.13.1), where we are sending current joint positions with the user defined torque command. 

Unfortunately, this only fixes the old callback style control in external mode and not the new active control mode. I am afraid this doesn't solve your problem @tkelestemur-bdai, but real fix is on the way.

BarisYazici, 2023-12-05:
The workaround is now also adapted to the `Active Control` in Libfranka - https://github.com/frankaemika/libfranka/tree/0.13.2. So if you update your libfranka to the newest version, you should not be getting this error anymore.

closed 2023-12-05 by BarisYazici

tkelestemur-bdai, 2023-12-05:
Thank you @BarisYazici, we'll try it asap.