frankarobotics/franka_ros2 #34: Frank ROS 2 support for FER and libfranka v0.9.2

MadsRossen, 2023-09-13:
Is there any plan for support of the discontinued Franka Emika Robot for ROS 2 humble,and it's compatible libfranka v0.9.2 ? Or am i just stuck with very expensive robot without a ROS 2 interface.

Tejas-Shastha, 2023-09-15:
I would also be interested in knowing this, as we have a few FER1s that we are very interested in actively using with ROS2 going forward.
As of now the [compatibility table shows](https://support.franka.de/docs/compatibility.html) that the FER1 does not receive the 5.2.0 firmware update -> which means it does not support libfranka 0.11.0 -> which means we cannot use the  humble branch of this driver.

drewskoots, 2023-09-16:
This   https://github.com/frankaemika/franka_ros2/pull/10 (now closed) pull request contains a franka_ros2 version compatible with Panda. It's effectively just a port of the older foxy franka_ros2, so it may not enable all the functionality of the official humble port.

Tejas-Shastha, 2023-09-18:
> This #10 (now closed) pull request contains a franka_ros2 version compatible with Panda. It's effectively just a port of the older foxy franka_ros2, so it may not enable all the functionality of the official humble port.

@drewskoots Thanks for the tip! This helps the interim to have even a slightly outdated code base to work with existing hardware.

However, it would be very helpful for us to know what the long term support plan for the Panda (FER1) is w.r.t ROS2, even beyond humble.

andrejpan, 2023-09-20:
This could be a bit difficult in the current situation.
https://www.linkedin.com/posts/frankaemika_franka-emika-news-activity-7102968220122304513-AfYd?utm_source=share&utm_medium=member_desktop

MadsRossen, 2023-09-27:
> @drewskoots Thanks for the tip! This helps the interim to have even a slightly outdated code base to work with existing hardware.
>
> However, it would be very helpful for us to know what the long term [...]

Thanks for the Tip!  I tried to run the different examples from the documentation using pull request #10 , the only one that does not work seems to be the following command: 
```
ros2 launch franka_moveit_config moveit.launch.py robot_ip:=<fci-ip>
```
When the robot is moved to another position using the ```MotionPlanning```, the robot does not hold it's position. I think it is a problem with the ```JointTrajectoryController```.

Tejas-Shastha, 2023-09-27:
> Thanks for the Tip! I tried to run the different examples from the documentation using pull request #10 , the only one that does not work seems to be the following command:
>
> ```
> ros2 launch [...]

Try combining libfranka `0.9.2` with this ros2 driver's version `v0.1.0`, this works for me on humble, to even run moveit with the robot.

tenfoldpaper, 2023-10-12:
I've just finished developing the first version of [panda_ros2](https://github.com/tenfoldpaper/panda_ros2), which is a port of franka_ros into ROS2 for the FER robots. It's been tested with libfranka 0.9.2 and Panda ver. 4.2.2. Most features for one arm scenario are working, though more thorough testing is necessary. Still, I think it's an improvement over what we have for the Panda robots in ROS2 at the moment. 
I would appreciate it if the community could try it out and let me know about bugs or if anything is missing!

BarisYazici, 2023-10-12:
> I've just finished developing the first version of [panda_ros2](https://github.com/tenfoldpaper/panda_ros2), which is a port of franka_ros into ROS2 for the FER robots. It's been tested with [...]

Looks really good! Good job! The feature that causes incompatibility is the ActiveControl that introduced with https://github.com/frankaemika/libfranka/releases/tag/0.11.0 . ActiveControl ensures that libfranka control loop and ROS 2 hardware interface run in the same thread. You can potentially port the ActiveControl feature in Libfranka fr3-develop branch to FER branch, and you would be able to use the franka_ros2 with libfranka for FER that has external control feature.

tenfoldpaper, 2023-10-16:
> Looks really good! Good job! The feature that causes incompatibility is the ActiveControl that introduced with https://github.com/frankaemika/libfranka/releases/tag/0.11.0 . ActiveControl ensures [...]

Thanks! I will take a look at it, after implementing the multi-arm feature.

yilmazabdurrah, 2024-01-22:
> I've just finished developing the first version of [panda_ros2](https://github.com/tenfoldpaper/panda_ros2), which is a port of franka_ros into ROS2 for the FER robots. It's been tested with [...]

Please note that it is also working with libfranka 0.8.0 and Panda FCI 4.0.1.

tingelst, 2024-02-05:
> Looks really good! Good job! The feature that causes incompatibility is the ActiveControl that introduced with https://github.com/frankaemika/libfranka/releases/tag/0.11.0 . ActiveControl ensures [...]

For those interested, I have now been able to run the `humble` branch on an FER (v4.2.2) by modifying `libfranka` to employ v0.9.2 `libfranka-common`. You can find the updates at https://github.com/tingelst/libfranka, see the `active-control-backport` branch. 

The effort and velocity controllers work, but the position controllers triggers a `joint_motion_generator_acceleration_discontinuity`.

BarisYazici, 2024-02-05:
@tingelst Good one. FR3 has a better communication architecture that could be the reason you are receiving `joint_motion_generator_acceleration_discontinuity` errors. 

> The effort and velocity controllers work, but the position controllers triggers a joint_motion_generator_acceleration_discontinuity.

You could try to activate rate limiter for the joint position controller [rate_limiter](https://github.com/frankaemika/franka_ros2/blob/40cf08b6f0c2d66fa568b2a3336f1ac7971995ed/franka_hardware/include/franka_hardware/robot.hpp#L302)

peterdavidfagan, 2024-04-01:
@tingelst should one be building from your latest branch? I am attempting to upgrade our libfranka version based on your fork but I am encountering build errors.

peterdavidfagan, 2024-04-01:
Ok pr is opened [here](https://github.com/tingelst/libfranka/pull/1) in case others are also building on @tingelst's work. Thanks again @tingelst for backporting and @BarisYazici for supporting.

peterdavidfagan, 2024-04-02:
I am in the process of testing the backport on my own [repository](https://github.com/peterdavidfagan/ros2_robotics_research_toolkit). It seems when I build libfranka from source where I set the robot and gripper version to be compatible with my control server and robot, everything builds but I get the following output from the `franka_bringup.launch.py`:

```
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2024-04-02-10-12-59-523989-r2d2-48611
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [48614]
[INFO] [joint_state_publisher-2]: process started with pid [48616]
[INFO] [ros2_control_node-3]: process started with pid [48618]
[INFO] [spawner-4]: process started with pid [48620]
[INFO] [spawner-5]: process started with pid [48622]
[INFO] [rviz2-6]: process started with pid [48624]
[robot_state_publisher-1] [INFO] [1712049179.798216960] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-1] [INFO] [1712049179.798274499] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-1] [INFO] [1712049179.798279552] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-1] [INFO] [1712049179.798282690] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-1] [INFO] [1712049179.798286068] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-1] [INFO] [1712049179.798289056] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-1] [INFO] [1712049179.798291685] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-1] [INFO] [1712049179.798294266] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-1] [INFO] [1712049179.798296810] [robot_state_publisher]: got segment panda_link8
[ros2_control_node-3] [WARN] [1712049179.803412685] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1712049179.803569966] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1712049179.806129705] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1712049179.806160578] [FrankaHardwareInterface]: Connecting to robot at "192.168.106.99" ...
[joint_state_publisher-2] [INFO] [1712049180.096138470] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[rviz2-6] [INFO] [1712049181.010636266] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1712049181.010739051] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-6] [INFO] [1712049181.022895012] [rviz2]: Stereo is NOT SUPPORTED
[spawner-4] [INFO] [1712049182.077244600] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-5] [INFO] [1712049182.089841105] [spawner_franka_robot_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-4] [INFO] [1712049184.083718853] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-5] [INFO] [1712049184.096185431] [spawner_franka_robot_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-4] [INFO] [1712049186.090269266] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-5] [INFO] [1712049186.102365654] [spawner_franka_robot_state_broadcaster]: Waiting for '/controller_manager' services to be available
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-6] [INFO] [1712049187.575781400] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1712049187.575798469] [rclcpp]: signal_handler(signum=2)
[ros2_control_node-3] [INFO] [1712049187.575840505] [rclcpp]: signal_handler(signum=2)
[spawner-5] Traceback (most recent call last):
[spawner-5]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-5]     sys.exit(load_entry_point('controller-manager==2.40.0', 'console_scripts', 'spawner')())
[spawner-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-5]     if not wait_for_controller_manager(
[spawner-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 117, in wait_for_controller_manager
[spawner-5]     return wait_for_value_or(
[spawner-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 66, in wait_for_value_or
[spawner-5]     time.sleep(0.2)
[spawner-5] KeyboardInterrupt
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-4]     sys.exit(load_entry_point('controller-manager==2.40.0', 'console_scripts', 'spawner')())
[spawner-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-4]     if not wait_for_controller_manager(
[spawner-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 117, in wait_for_controller_manager
[spawner-4]     return wait_for_value_or(
[... cut ...]
[ERROR] [ros2_control_node-3]: process has died [pid 48618, exit code -9, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_8wet03uw --params-file /home/robot/Code/test_franka/install/franka_bringup/share/franka_bringup/config/controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-3] was required: shutting down launched system
```

I am still debugging why this is occuring but @BarisYazici if you are aware of a reason why I could expect this to occur your advice would be appreciated. I was hoping the backport for libfranka would work out of the box with the humble branch of `franka_ros2`.

tingelst, 2024-04-02:
I checked on our Franka again today with the libfranka you can find [here](https://github.com/tingelst/libfranka/files/14834812/libfranka-active-control.zip) and installed by:

```bash
sudo dpkg -i libfranka-0.9.3-x86_64.deb
```

For `franka_ros2` to compile you have to change the required `libfranka` version in `franka_hardware`, `franka_gripper`, and `franka_semantic_components` to `0.9.3` as can be seen in my fork on the `fer` branch: https://github.com/tingelst/franka_ros2/

I did not encounter any problems running:
```bash
ros2 launch franka_bringup franka.launch.py robot_ip:=192.168.106.99 use_rviz:=true load_gripper:=false
```

@peterdavidfagan: Feel free to reach out for a meeting if you encounter any problems getting it to work.

peterdavidfagan, 2024-04-02:
Thanks @tingelst, this is much appreciated I will test the above on my repo. I'm also going to start investing more time in understanding the libfranka codebase so would be open to help maintain future backports based on changes made upstream (if/when possible).

peterdavidfagan, 2024-04-03:
Latest version posted by @tingelst looks to work, thank you so much @tingelst.

peterdavidfagan, 2024-04-03:
Thanks for swift responses, I will add to my task list the item of opening a pr to the official repos for this backport. I'll be sure to tag you both in the pr and attribute commits appropriately in the pr.

BarisYazici, 2024-04-03:
I cannot guarantee a backport. But feel free to start a PR.

Zarnack, 2024-04-18:
> @tingelst Good one. FR3 has a better communication architecture that could be the reason you are receiving `joint_motion_generator_acceleration_discontinuity` errors.
>
>
> You could try to activate [...]

Hi,
I just tried the backport from @tingelst and the backport seems to be working successfully. Unfortunately I am only able to use the effort command-interface.
Using the velocity interface I instantly get an "joint_motion_generator_acceleration_discontinuity" error at startup. Using the position interface I get the same error + "joint_motion_generator_velocity_discontinuity" as soon as I want to execute a movement.

Activating the rate limiter and the low pass filter does not change this behavior.

Am I am missing something? Why is this error even occurring when using the rate limiter?

I am using the FER without the franka gripper and disabled it in the webinterface.

peterdavidfagan, 2024-04-26:
Hi @Zarnack, 

I can also look to test the above on my setup at some point. So far I have only been using the effort interface.

Zarnack, 2024-06-05:
Hi @peterdavidfagan, have you maybe had some time testing the above on your setup?

peterdavidfagan, 2024-06-05:
Hi @Zarnack, I am still using the effort interface in my setup and its working for my use cases without issues. I plan to test new controllers soon, but again these all use the effort interface. I am keen to test the other interfaces but haven't done so yet.

Zarnack, 2024-06-05:
Ok. Thanks for the quick response!

christian-rauch, 2025-01-15:
If you want to use the ROS 2 node with the upstream `libfranka` 0.9.2 version on `humble` or `jazzy`, there is also https://github.com/boschresearch/franka_ros2/tree/fer1/.

AntoSave, 2025-02-24:
> For those interested, I have now been able to run the `humble` branch on an FER (v4.2.2) by modifying `libfranka` to employ v0.9.2 `libfranka-common`. You can find the updates at [...]

After several hours of investigating the issue you faced with position controllers I still couldn't manage to find a solution. 

The strange thing is that [this libfranka demo of an external joint position control loop](https://github.com/tingelst/libfranka/blob/active-control-backport/examples/generate_joint_position_motion_external_control_loop.cpp) works fine on my FER, and the implementation of the hardware interface in `franka_ros2` v0.1.15 (the version I am currently using) is basically equivalent to the demo code. 

To summarize some of the experiments I did, I tried to:
- set the collision behavior and the joint/cartesian impedance before starting the control loop as in the demo;
- change the `controller_manager` loop frequency to frequencies higher than and lower than the default 1000 Hz;
- use the exact same position reference from the demo starting from the same initial position;
all of this **without activating the `rate_limiter`** (as in the working demo), but I had no luck.

> [@tingelst](https://github.com/tingelst) Good one. FR3 has a better communication architecture that could be the reason you are receiving `joint_motion_generator_acceleration_discontinuity` errors.

I don't think that's the cause of the problem as the demo works fine, but please correct me if I'm wrong.

> You could try to activate rate limiter for the joint position controller [...]

I did some test runs with the rate_limiter without success. However, in libfranka v0.15.0 (from which the backporting was obtained) the FR3 constraints are hardcoded in the limiter, hence I should try by manually changing them to those of the FER.

@BarisYazici @tingelst Given my observations, do you have any ideas on what could be the cause of the problem?