frankarobotics/franka_ros2 #36: Humble fake controller moveit partially broken

Tejas-Shastha, 2023-09-15:
Testing within the dev container by following the instructions [here](https://support.franka.de/docs/franka_ros2.html), planning works, execution seems to work but does not really work:

* I was able to get into the dev container, build and launch the fake moveit instance. 
* Rviz popper up, I planned to a valid goal, and hit execute. 
* The console in VSCode showed positive feedback and eventually execution success after some seconds as expected.
* The MotionPlanning visualization in rviz also showed execution success after some seconds
* But the robot did not actually move. This can be verified by introspecting the robot state visually, tf tree etc..

BarisYazici, 2023-09-26:
I am guessing the issue might be that the MoveIt with JTC currently only using effort command interface and that's only changing the effort values in the fake hardware. So the fake hardware's position values are never changing. This could be potentially solved when we support `position` command interface.

Tejas-Shastha, 2023-09-27:
> I am guessing the issue might be that the MoveIt with JTC currently only using effort command interface and that's only changing the effort values in the fake hardware. So the fake hardware's position values are never changing.
> 
> This could be potentially solved when we support joint `position` command interface. Then the JTC can directly command the joint position values to the fake hardware which would make the robot actually move.

Yes indeed I came to a similar conclusion a few days ago after some brief experimentation as well. As you say, only the position command interface works with the fake controllers for the demo (which also requires adding this into the URDF), and only the effort command interface works with the robot. They are mutually exclusive and cannot both be defined simultaneously. I would imagine an elegant solution probably means propagating the `use_fake_hardware` flag a bit deeper to switch between these contexts.

BarisYazici, 2023-09-27:
Joint position command interface will be supported soon. But until then if you wanna use fake hardware, you could add the position command interface in the URDF file and change the interface in this file https://github.com/frankaemika/franka_ros2/blob/humble/franka_moveit_config/config/panda_ros_controllers.yaml#L21

BarisYazici, 2024-10-31:
Position interface is already added to the URDF. So we support joint level position commands. It's here https://github.com/frankaemika/franka_description/blob/d722358a418126296a7ad0d79a53c2ba7f6da46c/robots/common/franka_arm.ros2_control.xacro#L41
If you are having trouble to move the fake moveit robot you need to change this line to use tell moveit to use position interface. https://github.com/frankaemika/franka_ros2/blob/86ff6f49b1a1f5eb3c7ac5ae04e582df57a0022e/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml#L21. Beware this will not work with the real hardware robot. Only use this for the fake hardware moveit.

closed 2024-10-31 by BarisYazici