frankarobotics/franka_ros2 #39: Errors building franka_ros2 in ros Iron

mikelasa, 2023-10-10:
Hello,

Because of circunstances I have ros Iron installed and I tried to install franka_ros2 even if humble is the recommended one. So, I followed the steps in the tutorial adapting the steps to iron:
[https://support.franka.de/docs/franka_ros2.html](url)

```
sudo apt install -y libpoco-dev libeigen3-dev
git clone https://github.com/frankaemika/libfranka.git --recursive
cd libfranka
git switch 0.11.0
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF  ..
cmake --build . -j$(nproc)
cpack -G DEB
sudo dpkg -i libfranka-0.9.2-x86_64.deb 

sudo apt install -y ros-iron-hardware-interface ros-iron-generate-parameter-library ros-iron-ros2-control-test-assets ros-iron-controller-manager ros-iron-control-msgs ros-iron-xacro ros-iron-angles ros-iron-ros2-control ros-iron-realtime-tools ros-iron-control-toolbox ros-iron-moveit ros-iron-ros2-controllers ros-iron-joint-state-publisher ros-iron-joint-state-publisher-gui ros-iron-ament-cmake ros-iron-ament-cmake-clang-format python3-colcon-common-extensions 

mkdir -p ~/franka_ros2_ws/src

source /opt/ros/iron/setup.bash
cd ~/franka_ros2_ws
git clone https://github.com/frankaemika/franka_ros2.git src/franka_ros2
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.sh
```
But somehow fails in joint_trayectory_controller and aborts franka_msgs. Can someone help me with this issue?

```
`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.sh
Cloning into 'src/franka_ros2'...
remote: Enumerating objects: 2149, done.
remote: Counting objects: 100% (692/692), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 2149 (delta 583), reused 585 (delta 525), pack-reused 1457
Receiving objects: 100% (2149/2149), 4.06 MiB | 2.01 MiB/s, done.
Resolving deltas: 100% (1262/1262), done.
[1.518s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'joint_trajectory_controller' is in: /opt/ros/iron
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding joint_trajectory_controller

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> franka_msgs
Starting >>> franka_description
Starting >>> integration_launch_testing                    
Starting >>> joint_trajectory_controller
Finished <<< integration_launch_testing [0.91s]                  
Finished <<< franka_description [1.18s]                          
--- stderr: joint_trajectory_controller                                
In file included from /home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:37:
/opt/ros/iron/include/rclcpp/rclcpp/qos_event.hpp:18:2: warning: #warning This header is obsolete, please include rclcpp/event_handler.hpp instead [-Wcpp]
   18 | #warning This header is obsolete, please include rclcpp/event_handler.hpp instead
      |  ^~~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::CallbackReturn joint_trajectory_controller::JointTrajectoryController::on_configure(const rclcpp_lifecycle::State&)’:
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:685:26: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘desired’
  685 |   state_publisher_->msg_.desired.positions.resize(dof_);
      |                          ^~~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:686:26: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘desired’
  686 |   state_publisher_->msg_.desired.velocities.resize(dof_);
      |                          ^~~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:687:26: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘desired’
  687 |   state_publisher_->msg_.desired.accelerations.resize(dof_);
      |                          ^~~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:688:26: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘actual’
  688 |   state_publisher_->msg_.actual.positions.resize(dof_);
[... 17 lines cut ...]
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:915:28: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘actual’
  915 |     state_publisher_->msg_.actual.positions = current_state.positions;
      |                            ^~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:919:30: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘actual’
  919 |       state_publisher_->msg_.actual.velocities = current_state.velocities;
      |                              ^~~~~~
/home/mikel/franka_ros2_ws/src/franka_ros2/joint_trajectory_controller/src/joint_trajectory_controller.cpp:924:30: error: ‘struct control_msgs::msg::JointTrajectoryControllerState_<std::allocator<void> >’ has no member named ‘actual’
  924 |       state_publisher_->msg_.actual.accelerations = current_state.accelerations;
      |                              ^~~~~~
gmake[2]: *** [CMakeFiles/joint_trajectory_controller.dir/build.make:76: CMakeFiles/joint_trajectory_controller.dir/src/joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:208: CMakeFiles/joint_trajectory_controller.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< joint_trajectory_controller [10.5s, exited with code 2]
Aborted  <<< franka_msgs [15.0s]                                   

Summary: 2 packages finished [16.3s]
  1 package failed: joint_trajectory_controller
  1 package aborted: franka_msgs
  1 package had stderr output: joint_trajectory_controller
  7 packages not processed
not found: "/home/mikel/franka_ros2_ws/install/joint_trajectory_controller/share/joint_trajectory_controller/local_setup.sh"
`
```

BarisYazici, 2023-12-05:
maybe try adding this line to the build argument `--allow-overriding joint_trajectory_controller`. Be aware we have not tested the packages with Iron, only Humble distro is officially supported.

AndreasKuhner, 2024-10-30:
Hi @mikelasa ,
did it work in the end? If not, please re-open the issue and find a solution :smile:

closed 2024-10-30 by AndreasKuhner