frankarobotics/franka_ros2 #40: Cartesian controllers

jsll, 2023-10-12:
Hi,

I would like to have cartesian control of the Franka robot. Unfortunately, no cartesian position or velocity controllers are implemented. 

Are there any plans to release such controllers soon? If not, could someone point out what I would need to do to implement a cartesian position controller myself?

All the best,
Jens

BarisYazici, 2023-10-25:
Cartesian command interfaces are in progress. Will be available soon.

BarisYazici, 2023-11-13:
Cartesian velocity is implemented and soon the Cartesian Pose command interface will be ready.

BarisYazici, 2023-12-05:
Cartesian pose command interface is now supported in [v0.1.9](https://github.com/frankaemika/franka_ros2/tree/v0.1.9)

closed 2023-12-05 by BarisYazici