frankarobotics/franka_ros2 #44: Using the internal Impedance controllers through ROS2

Kzernobog, 2023-12-08:
Hi, 

This might be a noob question.

The `libfranka` interface allows one to generate their own motions and then speak to the Robot control using the `motion_generator` interface and a default option of either the internal joint Impedance controller or the internal cartesian Impedance controller. 

Is there a way of utilising the internal controllers through ROS2 api as well?

BarisYazici, 2023-12-08:
Hi,

It's not a noob question you already now about the motion_generators, which is advanced :) 

In ROS 2, we communicate also through the FCI(libfranka) interface with the Robot. You can think of ROS 2 as a wrapper of libfranka. In libfranka you can implement your control algorithm in two ways. First is the old callback style control, seconds is the  relatively new one available only for new FR3 robot is the `Active Control`, which readOnce, writeOnce interface. 

Motion_generators are using old callback function style control.  Like in this [callback function](https://github.com/frankaemika/libfranka/blob/3fe9b0853f4be58258814fb69921e5c59e81a9c5/examples/generate_joint_position_motion.cpp#L47) in the examples. We define the callback function, which takes `robot_state` and `period` as a argument and returns `JointPositions` based on some period function. And this callback function can return command interfaces types we support: `JointPositions`, `JointVelocites`, `JointTorques`, `CartesianVelocity`, `CartesianPose`. As you mentioned you can also pass the internal control type(JointImpendace, CartesianImpedance) to this [control function in the old callback style code](https://github.com/frankaemika/libfranka/blob/3fe9b0853f4be58258814fb69921e5c59e81a9c5/include/franka/robot.h#L320).

Now for the second approach with `Active Control`, the control style we support in ROS 2, is primarily developed to avoid two different threads in ROS 2 control package. This control style is simply usable as below.

``` cpp
auto active_control = startJointVelocityControl(franka::JointImpedance);
auto [robot_state, _] = active_control.readOnce();
active_control.writeOnce(joint_velocities);
```
In ROS 2, we currently defaulted the usage of internal control to franka::JointImpedance (https://github.com/frankaemika/franka_ros2/blob/4cea20feee12b2403bc6a7d2eaee98801e048f8f/franka_hardware/src/robot.cpp#L258). But you can easily change that to the franka::CartesianImpedance and compile again. Soon we can make it also configurable as in ROS 1.

BarisYazici, 2023-12-08:
You are using the internal controller, unless you are commanding torques to the robot, which external control.

You cannot change the stiffness or damping when a controller is running. You need to first stop the controller set the stiffness and activate the controller again