frankarobotics/franka_ros2 #45: Cartesian Impedance controller implementation

Kzernobog, 2023-12-12:
Hi,

I was wondering if there was someone already working on porting the cartesian impedance controller to ROS2. If not, could I give it a shot?

Thanks in advance

BarisYazici, 2023-12-18:
sure, give it a shot :)

danielsanjosepro, 2025-08-14:
hey! this is an old issue but you might still be interested: we have been working on a cartesian impedance ros2 controller, which is robot agnostic. In our lab we use it with the FR3 and it works pretty good. We have a few [dockerized demos](https://github.com/utiasDSL/crisp_controllers_demos) to try out the controller with the FR3:
```bash
docker compose build launch_franka
docker compose up launch_franka
```
We also have a small python interface and gym environment that can be used to train VLAs or other learning based models using LeRobot.
You can check out our [project website](https://utiasdsl.github.io/crisp_controllers/) which documents everything quite well :)

kamiradi, 2025-08-14:
This is very impressive. Thanks for open sourcing

closed 2025-08-14 by kamiradi