frankarobotics/franka_ros2 #48: model_example_controller crashes in simulation

RodrigoMA170393, 2024-01-12:
Currently, I was trying to visualize the dynamic parameters of the robot using the command

ros2 launch franka_bringup model_example_controller.launch.py robot_ip:=dont-care use_fake_hardware:=true use_rviz:=true

But I get the following output:

[image]
I think that it interprets the robot name as a state and it gives an error.
I am using libfranka 0.13.2 and franka_ros2 0.1.11.

BarisYazici, 2024-01-13:
that wouldn't work with fake hardware. It needs to connect to the robot hardware to get the model parameters

AndreasKuhner, 2025-03-05:
If there is any other issue, please let us know :)

closed 2025-03-05 by AndreasKuhner