frankarobotics/franka_ros2 #50: Control Error & Reflex Recovery

jcarpinelli-bdai, 2024-03-12:
The [`franka_ros`](https://support.franka.de/docs/franka_ros.html?highlight=reflex#franka-control) documentation specifies an `ErrorRecoveryAction` which can be used to restart controller nodes after an error or reflex. **Is there a similar mechanism in `franka_ros2`?** I do not see any mention of error recovery in the [ROS2 documentation](https://support.franka.de/docs/franka_ros2.html).

BarisYazici, 2024-03-26:
Good point. It's not yet available in Franka_ros2. We will prepare it for the coming releases.

jcarpinelli-bdai, 2024-03-26:
> Good point. It's not yet available in Franka_ros2. We will prepare it for the coming releases.

Thank you! For what it's worth... I think it would be really valuable to have an allow-list of errors that we want `franka_hardware` to ignore. In ROS1, these errors delay the control loop because they are thrown, then caught. We see the robot "hiccup" as the error is caught on the user side. It would be helpful if we could prevent `franka_hardware` from throwing some errors in the first place. In addition, it would be helpful to have more control over the thresholds for some errors. The 20ms UDP timeout in particular would be great to have the option to increase.

jcarpinelli-bdai, 2024-07-08:
@BarisYazici do you all have an ETA on this? I may have limited availability to help add error recovery to these `hardware_interface` classes.

AndreasKuhner, 2025-03-05:
Was merged some time ago 🥳

closed 2025-03-05 by AndreasKuhner