frankarobotics/franka_ros2 #53: Cartesian Stiffness Values Ignored

jcarpinelli-bdai, 2024-03-29:
When I use the Cartesian pose interface through `franka_semantic_components`, the arm seems to ignore the Cartesian stiffness values I set through the `/service_center`. The Joint impedance values are respected, but no matter how low I set the Cartesian stiffness values, the arm is not compliant in Cartesian space. 

To replicate this, you can add the following code to the `on_configure` method of the `CartesianImpedanceExampleController` class. **Am I missing some way to configure the arm for compliant motion in Cartesian space?**

```cpp
  auto client =
      get_node()->create_client<franka_msgs::srv::SetCartesianStiffness>("service_server/set_cartesian_stiffness");
  
  auto request = std::make_shared<franka_msgs::srv::SetCartesianStiffness::Request>(franka_msgs::srv::SetCartesianStiffness::Request());
  request->cartesian_stiffness.at(0) = 10;
  request->cartesian_stiffness.at(1) = 10;
  request->cartesian_stiffness.at(2) = 10;
  request->cartesian_stiffness.at(3) = 10;
  request->cartesian_stiffness.at(4) = 10;
  request->cartesian_stiffness.at(5) = 10;

  auto future_result = client->async_send_request(request);
  future_result.wait_for(seconds(1));

  auto success = future_result.get();
  if (!success) {
    RCLCPP_FATAL(get_node()->get_logger(), "Failed to set Cartesian stiffness.");
    return controller_interface::CallbackReturn::ERROR;
  } else if (!success->success) {
    RCLCPP_FATAL(get_node()->get_logger(), "Failed to set Cartesian stiffness.");
    return controller_interface::CallbackReturn::ERROR;
  } else {
    RCLCPP_INFO(get_node()->get_logger(), "Cartesian stiffness set.");
  }
```

AndreasKuhner, 2024-04-03:
This seems to be a small oversight of us - currently there is no way to do CartesianImpedance control (which then would re-act to the Cartesian stiffnesses you set).

But: You can work around it by changing in line 293 to 303 in https://github.com/frankaemika/franka_ros2/blob/humble/franka_hardware/src/robot.cpp to a `research_interface::robot::Move::ControllerMode::kCartesianImpedance` (or give the initialization function a parameter to be either/or)

AndreasKuhner, 2026-11-12:
Closing this issue as answered above:

1. **Internal FCI motion generators (`FrankaCartesianPoseInterface` / `FrankaCartesianVelocityInterface`):** `franka_hardware` starts libfranka's motion generators in `ControllerMode::kJointImpedance` (`franka_hardware/src/robot.cpp`), where compliance is governed by `set_joint_stiffness` unless `kCartesianImpedance` is selected.
2. **Torque-level Cartesian Impedance in ROS 2:** On `jazzy`, `franka_example_controllers` also includes `cartesian_impedance_example_controller`, which computes Cartesian and nullspace impedance directly on the joint effort interfaces and exposes `translational_stiffness`, `rotational_stiffness`, and `nullspace_stiffness` as ROS parameters.

closed 2026-11-12 by AndreasKuhner