frankarobotics/franka_ros2 #56: Parameterized Launch Configurations

jcarpinelli-bdai, 2024-04-17:
Right now, the `franka_bringup` package has launch files which load a static `controllers.yaml` file, which has all of the `*_example_controller` controllers. I think it would be helpful to have a parameter that _defaults_ to the `controllers.yaml` file in the `franka_bringup` package. This way, users could benefit from using `franka_bringup`, while still loading their own controllers.

If this capability isn't added, I think users _have_ to execute `ros2 param set` commands as shell processes, either in launch files or manually in a terminal.  

@BarisYazici do you agree that a `controller_manager_parameters` parameter, which defaults to `controllers.yaml`, is acceptable? If so, I can submit a PR with this change.

AndreasKuhner, 2025-03-05:
Hi @jcarpinelli-bdai ,
if you still have the commit, we could have a look into it :)

Cheers,
Andreas

jcarpinelli-bdai, 2025-03-05:
We are using our own launch file! I think that's appropriate, no need for a PR (I don't think).

jcarpinelli-bdai, 2025-03-05:
I think we can close this!

closed 2025-03-10 by AndreasKuhner