frankarobotics/franka_ros2 #60: Controller Development

peterdavidfagan, 2024-04-29:
Hi Franka Emika Development Team,

I am in the process of developing controllers based on the examples given in this repository. In order to test my controller software I planned to run the software against a MuJoCo simulation of my robot. I have this integration and example controllers running in MuJoCo however, I am facing issue in getting the example controllers stable in MuJoCo. 

I had the following question resulting from this that I was hoping it might be possible to get advice on. I wished to ask when developing controllers for the FER and/or FR3, is there a recommended testing protocol to ensure the safety of the hardware (outside of existing safety checks in the software). When it comes to tuning controller parameters on both platforms are there any guides/advice the development team would provide on this?

BarisYazici, 2024-06-27:
Hey Peter,
If it helps you, we just released a Gazebo package. The example controllers are working with the Gazebo plugin. You can take a look at them.
I haven't tried the MuJoCo integration. But keep in mind that the example controller commanding torque is sending the torque without the gravity. You need to add the gravity term in the MuJoCo simulation section. You can take a look at how we feed the gravity torque in the Gazebo controller https://github.com/frankaemika/franka_ros2/blob/humble/franka_gazebo/franka_ign_ros2_control/src/ign_system.cpp#L714

peterdavidfagan, 2024-07-06:
This is super useful, thanks @BarisYazici.

BarisYazici, 2024-07-06:
If you get something ready, we could make it available to public.

peterdavidfagan, 2024-07-07:
> If you get something ready, we could make it available to public. So more developers can utilise it.

Right now my code is more of a proof of concept that I use for occasional debug. Right now I am more focused on writing and theory work but once I am back to software development work and I have the capacity to I'll clean up the code and post an example here. 

Also one of the MoveIt GSoC projects this year is working on a MuJoCo integration for ROS 2 control. I think will be worth checking out when it is completed.