frankarobotics/franka_ros2 #61: cartesian_pose_example_controller Issue

ALICEYZ5, 2024-05-08:
Hi franka scientist,
I was running **cartesian_pose_example_controller** in _franka_example_controllers_ (just ros2 launch franka_bringup cartesian_pose_example_controller.launch.py robot_ip:=<fclip>, without changing any code), but got the error: 
`libfranka: Move command aborted: motion aborted by reflex! ["cartesian_motion_generator_velocity_discontinuity", "cartesian_motion_generator_acceleration_discontinuity", "cartesian_motion_generator_joint_acceleration_discontinuity"]`
os: Ubuntu 22.04
ROS2 humble
franka_ros2 branch: humble, tag: v0.1.13
libfranka branch: fr3-develop, tag: 0.13.3
I was also trying to reduce the radius of motion, but no matter how much I reduced, the same error still show up, unless I set the radius - 0. besides, I tried to set cartesian_pose_command_rate_limit_active_ = true, does not help either.
Please advise.
Thanks,

olaghattas, 2024-11-06:
Any help with this issue? cartesian_pose_example_controller always gives  "motion aborted by reflex!" . When I run the example in libfranka (./examples/generate_cartesian_pose_motion ) it works but when I use ros2 it gives me the error. Is there a condition or anything that needs changing?

this is the command I am using "ros2 launch franka_bringup cartesian_pose_example_controller.launch.py arm_id:=fr3 robot_ip:="

AndreasKuhner, 2024-11-07:
Hi @olaghattas , 
what version of libfranka/franka_ros2 do you use? 

Cheers,
Andreas

ALICEYZ5, 2024-11-07:
Check your internet card. Franka website has requirement on the latency of
the card.

YE ZHANG
5953 High Point Ct., Brighton, MI 48116

ALICEYZ5, 2024-11-07:
Check your internet card. Franka website has requirement on the latency of
the card.

olaghattas, 2024-11-07:
Hey @AndreasKuhner, the libfranka version is 0.13.3  and franka_ros2 is v0.1.15. The compatibility sections show that they are compatible https://frankaemika.github.io/docs/compatibility.html

ALICEYZ5, 2024-11-07:
Please refer to this link for trouble shooting: https://frankaemika.github.io/docs/troubleshooting.html#network-bandwidth-delay-and-jitter-test

olaghattas, 2024-11-07:
@ALICEYZ5 Thanks for the suggestion, but I have already checked those steps before. 

I have included a more detailed description of what I am facing.

with ros2  running the communication_test command was successfully 
COMMAND:  communication_test  172.16.0.2 
OUTPUT:
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
Starting communication test.
#100 Current success rate: 1.00
#200 Current success rate: 1.00
#300 Current success rate: 1.00
#400 Current success rate: 1.00
#500 Current success rate: 1.00
#600 Current success rate: 1.00
#700 Current success rate: 1.00
#800 Current success rate: 1.00
#900 Current success rate: 1.00
#1000 Current success rate: 1.00
#1100 Current success rate: 1.00
#1200 Current success rate: 1.00
#1300 Current success rate: 1.00
#1400 Current success rate: 1.00
#1500 Current success rate: 0.98
#1600 Current success rate: 1.00

Also running it with libfranka was successful too
COMMAND:  ~/libfranka/build$ ./examples/communication_test 172.16.0.2
OUTPUT:  
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
#100 Current success rate: 1.00
#200 Current success rate: 1.00
#300 Current success rate: 1.00
#400 Current success rate: 1.00
#500 Current success rate: 1.00
#600 Current success rate: 1.00
#700 Current success rate: 1.00
#800 Current success rate: 1.00
#900 Current success rate: 1.00

When I run generate_cartesian_pose_motion from the libfranka repository it works fine and it moves the robot 
 
COMMAND: ~/libfranka/build$ ./examples/generate_cartesian_pose_motion 172.16.0.2
OUTPUT:  
 
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Finished motion, shutting down example

My problem is that when I try to run the same example but using the launch file in ros2 it generates an error however ros2 launch franka_bringup joint_velocity_example_controller.launch.py arm_id:=fr3 robot_ip:=172.16.0.2 doesn't 

COMMAND:  ros2 launch franka_bringup cartesian_pose_example_controller.launch.py arm_id:=fr3 robot_ip:=172.16.0.2

OUTPUT: 
[ros2_control_node-3] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-3]   what():  libfranka: Move command aborted: motion aborted by reflex! ["cartesian_motion_generator_velocity_discontinuity", "cartesian_motion_generator_acceleration_discontinuity"]
[ros2_control_node-3] Stack trace (most recent call last) in thread 9216:
[... 26 lines cut ...]
[ros2_control_node-3]       > 237:   const auto [current_state, _] = active_control_->readOnce();
[ros2_control_node-3]         238:   return current_state;
[ros2_control_node-3]         239: }
[ros2_control_node-3] #8    Object "/home/olagh/libfranka/build/libfranka.so.0.13.3", at 0x7f16e065b4d7, in franka::ActiveControl::readOnce()
[ros2_control_node-3] #7    Object "/home/olagh/libfranka/build/libfranka.so.0.13.3", at 0x7f16e0659d98, in franka::Robot::Impl::throwOnMotionError(franka::RobotState const&, unsigned int) [clone .cold]
[ros2_control_node-3] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f16e7dca4d7, in __cxa_throw
[ros2_control_node-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f16e7dca276, in std::terminate()
[ros2_control_node-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f16e7dca20b, in 
[ros2_control_node-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f16e7dbeb9d, in 
[ros2_control_node-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f16e7afb7f2]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f16e7b15475]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f16e7b699fc]
[ros2_control_node-3] Aborted (Signal sent by tkill() 9083 1000)
[ERROR] [ros2_control_node-3]: process has died [pid 9083, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /home/olagh/franka_ros2_ws/install/franka_bringup/share/franka_bringup/config/controllers.yaml --params-file /tmp/launch_params_fpog1o5n --params-file /tmp/launch_params_29mqhx8s -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-3] was required: shutting down launched system
[INFO] [franka_gripper_node-6]: sending signal 'SIGINT' to process[franka_gripper_node-6]
[INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
[INFO] [joint_state_publisher-1]: sending signal 'SIGINT' to process[joint_state_publisher-1]
[franka_gripper_node-6] [INFO] [1731001848.586929687] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1731001848.587340321] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 9081]
[INFO] [joint_state_publisher-1]: process has finished cleanly [pid 9079]
[INFO] [franka_gripper_node-6]: process has finished cleanly [pid 9089]

olaghattas, 2024-11-08:
updating to  0.13.6 didnt fix the issue.

Is  libfranka 0.14.2  compatible with 0.13.6, cause other people are working on the robot as well.

closed 2025-01-24 by BarisYazici