frankarobotics/franka_ros2 #62: No movement while using fake_hardware

luiscoelho23, 2024-05-16:
Hi,
I was trying to use the example controllers using fake_hardware but I get no movement on Rviz, changed the command interface on config\controllers.yaml to match the type of controller but still get no results.

Command exmaple:
ros2 launch franka_bringup joint_position_example_controller.launch.py robot_ip:=dont-care use_fake_hardware:=true use_rviz:=true

[image]

[image]

BarisYazici, 2024-07-18:
You could use the gazebo simulator examples instead. They should work

rraetz, 2024-08-19:
Hello everybody,

I get the same behavior with `ros2 launch franka_fr3_moveit_config moveit.launch.py use_fake_hardware:=true robot_ip:=yyy.yyy.yyy.yyy` The planning works but the faked robot just remains at its initial joint positions. 

Is there any solution to this? Or could anybody point me to where I need to look for bugs? I observed that on the `/joint_states` topic, the position does not change as the robot "moves" when executing a planned motion with the MoveIt plugin. However, the efforts change during motion execution. Any help is much appreciated.

BarisYazici, 2024-08-19:
> Hello everybody,
>
> I get the same behavior with `ros2 launch franka_fr3_moveit_config moveit.launch.py use_fake_hardware:=true robot_ip:=yyy.yyy.yyy.yyy` The planning works but the faked robot just [...]

If you change the command interface [here](https://github.com/frankaemika/franka_ros2/blob/humble/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml#L21) to position. That would solve your issue probably.

rraetz, 2024-08-22:
> If you change the command interface [here](https://github.com/frankaemika/franka_ros2/blob/humble/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml#L21) to position. That would solve your issue probably.

Awesome, this solved it! Thank you very much for your help.

closed 2024-10-07 by AndreasKuhner