frankarobotics/franka_ros2 #64: [bug] can't run `joint_position_example_controller` with ubuntu realtime kernel

TomCC7, 2024-06-11:
Hi, I'm not able to run the default `joint_position_example_controller` with ubuntu realtime kernel, but it's working when I'm using a non-realtime one. I'm not sure whether it will reproduce the error but here's the settings on my side:
+ Command I ran: `ros2 launch franka_bringup joint_position_example_controller.launch.py robot_ip:=172.16.0.2`
+ Kernel with no problem: 6.5.0-35-generic
+ Kernel with problem: 5.15.0-1063-realtime
+ `franka_ros2` version: https://github.com/frankaemika/franka_ros2/tree/846faec5b13d051311169f6c14543cd4a805ce1d
+ `libfranka` version: https://github.com/frankaemika/libfranka/tree/64b0314518ba28110e5559669a6bb310087ddc0c
+ arm firmware version: 5.6.0-33792dff37
+ arm version: franka production 3
+ CPU: Intel(R) Core(TM) i9-10900KF CPU @ 3.70GHz

Here's the error log:
```
[ros2_control_node-3] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-3]   what():  libfranka: Move command aborted: motion aborted by reflex! ["joint_motion_generator_velocity_discontinuity", "joint_motion_generator_acceleration_discontinuity"]
[ros2_control_node-3] Stack trace (most recent call last) in thread 13897:
[ros2_control_node-3] #18   Object "", at 0xffffffffffffffff, in
[ros2_control_node-3] #17   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f4eec90384f]
[ros2_control_node-3] #16   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f4eec871ac2]
[ros2_control_node-3] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f4eecb02252, in
[ros2_control_node-3] #14   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x562a936b2800, in
[ros2_control_node-3] #13   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f4eecf0b6f4, in controller_manager::ControllerManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f4eec71069c, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #11   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f4eec72de8f, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #10   Object "/workspace/codes/cc_control_ws/build/franka_hardware/libfranka_hardware.so", at 0x7f4ee57cb521, in franka_hardware::FrankaHardwareInterface::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #9    Object "/workspace/codes/cc_control_ws/build/franka_hardware/libfranka_hardware.so", at 0x7f4ee57ddc42, in franka_hardware::Robot::readOnce()
[ros2_control_node-3] #8    Object "/usr/lib/libfranka.so.0.13.3", at 0x7f4ee57234d7, in franka::ActiveControl::readOnce()
[ros2_control_node-3] #7    Object "/usr/lib/libfranka.so.0.13.3", at 0x7f4ee5721d98, in franka::Robot::Impl::throwOnMotionError(franka::RobotState const&, unsigned int) [clone .cold]
[ros2_control_node-3] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f4eecad44d7, in __cxa_throw
[ros2_control_node-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f4eecad4276, in std::terminate()
[ros2_control_node-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f4eecad420b, in
[ros2_control_node-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f4eecac8b9d, in
[ros2_control_node-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f4eec8057f2]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f4eec81f475]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f4eec8739fc]
[ros2_control_node-3] Aborted (Signal sent by tkill() 13840 1001)
[INFO] [spawner-5]: process has finished cleanly [pid 13844]
```

I've make sure that all the build are in release mode, so should have the best performance.

BarisYazici, 2024-07-31:
maybe the realtime kernel you have installed is not compatible with the drivers etc. If you have ubuntu 22.04, you can have a free individual pro account and that configures the real-time kernel compatible with your pc. https://ubuntu.com/real-time

AndreasKuhner, 2024-10-30:
Hi @TomCC7 ,
how is it going? Did you resolve the problem?

Cheers,
Andreas

closed 2024-10-30 by TomCC7