frankarobotics/franka_ros2 #65: CMake Cannot Find `boost_sml` Headers in ROS 2 Workspace

DimiSf, 2024-07-23:
# CMake Cannot Find `boost_sml` Headers in ROS 2 Workspace

I am encountering an issue with building the `franka_ign_ros2_control` package in my ROS 2 workspace. The build process fails because CMake cannot find the `boost_sml` headers. `boost_sml` is a header-only library from Boost and does not provide its own CMake configuration files.

I am using: 
- **Ubuntu 22.04**
- **CMake Version:** 11.10.2
- **ROS2:** Humble
- **BOOST_LIB_VERSION:** 1_74

## Error Message

CMake Error at CMakeLists.txt:24 (find_package):
By not providing "Findboost_sml.cmake" in CMAKE_MODULE_PATH this project
has asked CMake to find a package configuration file provided by
"boost_sml", but CMake did not find one.

Could not find a package configuration file provided by "boost_sml" with
any of the following names:
boost_smlConfig.cmake
boost_sml-config.cmake
Add the installation prefix of "boost_sml" to CMAKE_PREFIX_PATH or set
"boost_sml_DIR" to a directory containing one of the above files. If
"boost_sml" provides a separate development package or SDK, be sure it has
been installed.

## Current Setup

1. **Location of `boost_sml` Headers:**
   - The `sml.hpp` file is located at `/home/group/boost_sml/include/boost/sml.hpp`.

2. **Environment Variables:**
   - `CMAKE_PREFIX_PATH` includes `/home/group/boost_sml` and `/home/group/franka_ros2_ws/cmake`.

     ```sh
     echo $CMAKE_PREFIX_PATH
     /home/group/boost_sml:/home/group/franka_ros2_ws/cmake
     ```

3. **Custom CMake Module:**
   - Created `Findboost_sml.cmake` in `/home/group/franka_ros2_ws/cmake` with the following content:

     ```cmake
     # Findboost_sml.cmake
     # Locate the header files for boost_sml
     message(STATUS "CMAKE_PREFIX_PATH: ${CMAKE_PREFIX_PATH}")
     find_path(boost_sml_INCLUDE_DIRS
       NAMES boost/sml.hpp
       PATHS ${CMAKE_PREFIX_PATH}
       DOC "Path to the boost_sml header files"
     )

     if (boost_sml_INCLUDE_DIRS)
       message(STATUS "Found boost_sml: ${boost_sml_INCLUDE_DIRS}")
     else()
       message(FATAL_ERROR "Could not find boost_sml")
     endif()

     # Set the include directories
     set(boost_sml_INCLUDE_DIR ${boost_sml_INCLUDE_DIRS})

     # Mark the targets
     add_library(boost_sml INTERFACE)
     target_include_directories(boost_sml INTERFACE ${boost_sml_INCLUDE_DIR})
     ```

4. **CMakeLists.txt Configuration:**
   - Modified `CMakeLists.txt` for `franka_gazebo_bringup` and `franka_ign_ros2_control` to include the custom module path and find `boost_sml`.

   - **Example for `franka_gazebo_bringup`:**

     ```cmake
     cmake_minimum_required(VERSION 3.5)
     project(franka_gazebo_bringup)

     set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/home/group/franka_ros2_ws/cmake")

     find_package(ament_cmake REQUIRED)
     find_package(boost_sml REQUIRED)

     # Install project launch files
     install(
       DIRECTORY
         launch config
       DESTINATION share/${PROJECT_NAME}
     )

     if(BUILD_TESTING)
       find_package(ament_lint_auto REQUIRED)
       ament_lint_auto_find_test_dependencies()
     endif()

     ament_package()
     ```

   - **Example for `franka_ign_ros2_control`:**

     ```cmake
     cmake_minimum_required(VERSION 3.5)
     project(franka_ign_ros2_control)

     set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/home/group/franka_ros2_ws/cmake")

     find_package(ament_cmake REQUIRED)
     find_package(boost_sml REQUIRED)

     # Other configuration and dependencies...

     # Default to C11
     if(NOT CMAKE_C_STANDARD)
         set(CMAKE_C_STANDARD 11)
     endif()
     # Default to C++17
     if(NOT CMAKE_CXX_STANDARD)
         set(CMAKE_CXX_STANDARD 17)
     endif()

     # Compiler options
     if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
         add_compile_options(-Wall -Wextra -Wpedantic)
     endif()

     # Find dependencies
     find_package(ament_cmake REQUIRED)
     find_package(ament_index_cpp REQUIRED)
     find_package(controller_manager REQUIRED)
     find_package(hardware_interface REQUIRED)
     find_package(pluginlib REQUIRED)
     find_package(rclcpp REQUIRED)
     find_package(yaml_cpp_vendor REQUIRED)
     find_package(franka_hardware REQUIRED)
     find_package(urdf REQUIRED)
     find_package(kdl_parser REQUIRED)
     find_package(Franka 0.13.3 REQUIRED)
     find_package(tf2_geometry_msgs REQUIRED)
     find_package(tf2_eigen REQUIRED)

     # Ignition versions
     if("$ENV{IGNITION_VERSION}" STREQUAL "citadel")
       find_package(ignition-gazebo3 REQUIRED)
       set(IGN_GAZEBO_VER ${ignition-gazebo3_VERSION_MAJOR})
       message(STATUS "Compiling against Ignition Citadel")
     elseif("$ENV{IGNITION_VERSION}" STREQUAL "edifice")
       find_package(ignition-gazebo5 REQUIRED)
       set(IGN_GAZEBO_VER ${ignition-gazebo5_VERSION_MAJOR})
       message(STATUS "Compiling against Ignition Edifice")
     elseif("$ENV{IGNITION_VERSION}" STREQUAL "fortress")
       find_package(ignition-gazebo6 REQUIRED)
       set(IGN_GAZEBO_VER ${ignition-gazebo6_VERSION_MAJOR})
       message(STATUS "Compiling against Ignition Fortress")
     else()
       find_package(ignition-gazebo6 REQUIRED)
       set(IGN_GAZEBO_VER ${ignition-gazebo6_VERSION_MAJOR})
       message(STATUS "Compiling against Ignition Fortress")
     endif()

     find_package(ignition-plugin1 REQUIRED)
     set(IGN_PLUGIN_VER ${ignition-plugin1_VERSION_MAJOR})

     include_directories(include)

     add_library(${PROJECT_NAME}-system SHARED
       src/ign_ros2_control_plugin.cpp
       src/model_kdl.cpp
     )

     target_link_libraries(${PROJECT_NAME}-system
       ignition-gazebo${IGN_GAZEBO_VER}::core
       ignition-plugin${IGN_PLUGIN_VER}::register
     )
     ament_target_dependencies(${PROJECT_NAME}-system
       ament_index_cpp
       controller_manager
       hardware_interface
       pluginlib
       rclcpp
       yaml_cpp_vendor
       rclcpp_lifecycle
       franka_hardware
       Franka
       urdf
       kdl_parser
       tf2_geometry_msgs
       tf2_eigen
     )

     add_library(ign_hardware_plugins SHARED
       src/ign_system.cpp
       src/model_kdl.cpp
     )
     ament_target_dependencies(ign_hardware_plugins
       rclcpp_lifecycle
       hardware_interface
       rclcpp
       franka_hardware
       Franka
       urdf
       kdl_parser
       tf2_geometry_msgs
       tf2_eigen
     )
     target_link_libraries(ign_hardware_plugins
       ignition-gazebo${IGN_GAZEBO_VER}::core
     )

     # Install
     install(TARGETS
       ign_hardware_plugins
       ARCHIVE DESTINATION lib
       LIBRARY DESTINATION lib
       RUNTIME DESTINATION bin
     )

     install(DIRECTORY
       include/
       DESTINATION include
     )

     # Testing and linting
     if(BUILD_TESTING)
       find_package(ament_lint_auto REQUIRED)
       ament_lint_auto_find_test_dependencies()
     endif()

     ament_export_include_directories(include)
     ament_export_libraries(${PROJECT_NAME}-system ign_hardware_plugins)

     install(TARGETS ${PROJECT_NAME}-system
       DESTINATION lib
     )

     pluginlib_export_plugin_description_file(franka_ign_ros2_control ign_hardware_plugins.xml)

     ament_package()
     ```

To address the issue, steps were taken to ensure that `boost_sml` was correctly installed and detected by CMake. First, I verified that the `sml.hpp` file was present in the directory `/home/group/boost_sml/include/boost/sml.hpp`, confirming that `boost_sml` was installed. Next, I set the `CMAKE_PREFIX_PATH` environment variable to include both `/home/group/boost_sml` and `/home/group/franka_ros2_ws/cmake` by running the command:

```sh
export CMAKE_PREFIX_PATH=/home/group/boost_sml:/home/group/franka_ros2_ws/cmake

To further diagnose the issue, I added debug output to the Findboost_sml.cmake file to print the CMAKE_PREFIX_PATH and to provide status messages about the search for boost_sml. The modified Findboost_sml.cmake file included the following lines:

message(STATUS "CMAKE_PREFIX_PATH: ${CMAKE_PREFIX_PATH}")
find_path(boost_sml_INCLUDE_DIRS
  NAMES boost/sml.hpp
  PATHS ${CMAKE_PREFIX_PATH}
  DOC "Path to the boost_sml header files"
)

if (boost_sml_INCLUDE_DIRS)
  message(STATUS "Found boost_sml: ${boost_sml_INCLUDE_DIRS}")
else()
  message(FATAL_ERROR "Could not find boost_sml")
endif()

After making these changes, I cleaned the workspace by deleting the build, install, and log directories to ensure a fresh build environment. I then attempted to rebuild the workspace using the following commands:

cd /home/group/franka_ros2_ws
rm -rf build install log
colcon build

Despite these efforts, the build process still fails with the same error. Could someone please advise on what might be missing or incorrect in my setup? Any help would be greatly appreciated.

Thank you!

BarisYazici, 2024-07-31:
can you try to follow the steps with the devcontainer(https://frankaemika.github.io/docs/franka_ros2.html#use-vscode-devcontainer) and let us know if that works? All the dependency should be automatically installed under devcontainer

AndreasKuhner, 2024-10-30:
Hi @DimiSf ,
could you figure out the problem?

Cheers,
Andreas

DimiSf, 2024-10-30:
@BarisYazici  @AndreasKuhner  Hey there,thank you for reaching out! I had to erase everything and rebuild because i was very confused (i am a total beginner). So i didn't really  solved the issue about boost_sml ... but now i have moved on (till the next error).

AndreasKuhner, 2024-10-31:
Erasing everything and rebuilding is the expert trick #1 - wouldn't have done it differently :grin: 

If you have further question, just re-open the ticket or start a new one :smile:

closed 2024-10-31 by AndreasKuhner