frankarobotics/franka_ros2 #66: Problem with libfranka

MasterAufl, 2024-08-01:
I am attempting to use franka_ros2 version 0.1.6 with libfranka version 0.12.1. However, while building the workspace, I encounter an error stating that libfranka version 0.13.3 is required. According to the compatibility information provided [here](https://frankaemika.github.io/docs/compatibility.html), my current setup should be compatible. Am I overlooking something? Could you provide any advice on resolving this issue?

Best regards

MasterAufl, 2024-08-14:
I would like to use libfranka version 0.13.0. According to the compatibility table, franka_ros2 version 0.1.8 should work with it. However, when building the workspace, an error occurs stating that version 0.10.0 must be used. I have also tested the later versions of franka_ros2 and encountered the same error up to version 0.12.0. Starting from version 0.13.0, libfranka version 0.13.2 is required.

How can I use ROS2 with libfranka 0.13.0?
I would greatly appreciate any help!

AndreasKuhner, 2024-10-30:
Hi @MasterAufl ,
sorry for the late response. By any chance, did you already figure out a solution?

If not, what image version do you use at the moment? We highly recommend to update the image version to the nowadays 5.7.1 which should work with a libfranka 0.14.x.

And: As always, don't forget if you checkout another version of libfranka, to always do a `git submodule update` to also get the updated dependencies.