frankarobotics/franka_ros2 #70: Stumped at ["joint_velocity_violation"]

adamrankin, 2024-09-16:
I am working with a Franka Research 3 robot with OS version 5.6.0

I have installed ros2 humble on Ubuntu 22.04:
<details>
  <summary>
    Packages
  </summary>

```
ros-humble-ackermann-msgs/jammy,now 2.0.2-3jammy.20240728.214803 amd64 [installed,automatic]
ros-humble-ackermann-steering-controller/jammy,now 2.37.2-1jammy.20240823.151641 amd64 [installed,automatic]
ros-humble-action-msgs/jammy,now 1.2.1-1jammy.20240728.211412 amd64 [installed,automatic]
ros-humble-action-tutorials-cpp/jammy,now 0.20.5-1jammy.20240728.224822 amd64 [installed,automatic]
ros-humble-action-tutorials-interfaces/jammy,now 0.20.5-1jammy.20240728.211716 amd64 [installed,automatic]
ros-humble-action-tutorials-py/jammy,now 0.20.5-1jammy.20240728.221540 amd64 [installed,automatic]
ros-humble-actionlib-msgs/jammy,now 4.2.4-1jammy.20240728.214932 amd64 [installed,automatic]
ros-humble-actuator-msgs/jammy,now 0.0.1-1jammy.20240728.214936 amd64 [installed,automatic]
ros-humble-admittance-controller/jammy,now 2.37.2-1jammy.20240823.152956 amd64 [installed,automatic]
ros-humble-ament-clang-format/jammy,now 0.12.11-1jammy.20240728.200417 amd64 [installed,automatic]
ros-humble-ament-cmake-auto/jammy,now 1.3.10-1jammy.20240728.193007 amd64 [installed,automatic]
ros-humble-ament-cmake-clang-format/jammy,now 0.12.11-1jammy.20240728.200453 amd64 [installed]
ros-humble-ament-cmake-copyright/jammy,now 0.12.11-1jammy.20240728.200257 amd64 [installed,automatic]
ros-humble-ament-cmake-core/jammy,now 1.3.10-1jammy.20240728.190459 amd64 [installed,automatic]
ros-humble-ament-cmake-cppcheck/jammy,now 0.12.11-1jammy.20240728.200441 amd64 [installed,automatic]
ros-humble-ament-cmake-cpplint/jammy,now 0.12.11-1jammy.20240728.200446 amd64 [installed,automatic]
ros-humble-ament-cmake-export-definitions/jammy,now 1.3.10-1jammy.20240728.192015 amd64 [installed,automatic]
ros-humble-ament-cmake-export-dependencies/jammy,now 1.3.10-1jammy.20240728.192011 amd64 [installed,automatic]
ros-humble-ament-cmake-export-include-directories/jammy,now 1.3.10-1jammy.20240728.191515 amd64 [installed,automatic]
ros-humble-ament-cmake-export-interfaces/jammy,now 1.3.10-1jammy.20240728.191653 amd64 [installed,automatic]
ros-humble-ament-cmake-export-libraries/jammy,now 1.3.10-1jammy.20240728.191624 amd64 [installed,automatic]
ros-humble-ament-cmake-export-link-flags/jammy,now 1.3.10-1jammy.20240728.192014 amd64 [installed,automatic]
ros-humble-ament-cmake-export-targets/jammy,now 1.3.10-1jammy.20240728.191833 amd64 [installed,automatic]
ros-humble-ament-cmake-flake8/jammy,now 0.12.11-1jammy.20240728.200451 amd64 [installed,automatic]
ros-humble-ament-cmake-gen-version-h/jammy,now 1.3.10-1jammy.20240728.192636 amd64 [installed,automatic]
ros-humble-ament-cmake-gmock/jammy,now 1.3.10-1jammy.20240728.192653 amd64 [installed,automatic]
ros-humble-ament-cmake-gtest/jammy,now 1.3.10-1jammy.20240728.191907 amd64 [installed,automatic]
ros-humble-ament-cmake-include-directories/jammy,now 1.3.10-1jammy.20240728.192016 amd64 [installed,automatic]
ros-humble-ament-cmake-libraries/jammy,now 1.3.10-1jammy.20240728.191938 amd64 [installed,automatic]
ros-humble-ament-cmake-lint-cmake/jammy,now 0.12.11-1jammy.20240728.200131 amd64 [installed,automatic]
ros-humble-ament-cmake-pep257/jammy,now 0.12.11-1jammy.20240728.200531 amd64 [installed,automatic]
ros-humble-ament-cmake-pytest/jammy,now 1.3.10-1jammy.20240728.191915 amd64 [installed,automatic]
ros-humble-ament-cmake-python/jammy,now 1.3.10-1jammy.20240728.191604 amd64 [installed,automatic]
ros-humble-ament-cmake-ros/jammy,now 0.10.0-3jammy.20240728.202158 amd64 [installed,automatic]
ros-humble-ament-cmake-target-dependencies/jammy,now 1.3.10-1jammy.20240728.192104 amd64 [installed,automatic]
ros-humble-ament-cmake-test/jammy,now 1.3.10-1jammy.20240728.191740 amd64 [installed,automatic]
ros-humble-ament-cmake-uncrustify/jammy,now 0.12.11-1jammy.20240728.200425 amd64 [installed,automatic]
ros-humble-ament-cmake-version/jammy,now 1.3.10-1jammy.20240728.192011 amd64 [installed,automatic]
ros-humble-ament-cmake-xmllint/jammy,now 0.12.11-1jammy.20240728.200556 amd64 [installed,automatic]
ros-humble-ament-cmake/jammy,now 1.3.10-1jammy.20240728.192834 amd64 [installed]
ros-humble-ament-copyright/jammy,now 0.12.11-1jammy.20240728.194306 amd64 [installed,automatic]
ros-humble-ament-cppcheck/jammy,now 0.12.11-1jammy.20240728.191524 amd64 [installed,automatic]
ros-humble-ament-cpplint/jammy,now 0.12.11-1jammy.20240728.195948 amd64 [installed,automatic]
ros-humble-ament-flake8/jammy,now 0.12.11-1jammy.20240728.191938 amd64 [installed,automatic]
ros-humble-ament-index-cpp/jammy,now 1.4.0-2jammy.20240728.202210 amd64 [installed,automatic]
ros-humble-ament-index-python/jammy,now 1.4.0-2jammy.20240728.200002 amd64 [installed,automatic]
ros-humble-ament-lint-auto/jammy,now 0.12.11-1jammy.20240728.191921 amd64 [installed,automatic]
ros-humble-ament-lint-cmake/jammy,now 0.12.11-1jammy.20240728.200017 amd64 [installed,automatic]
ros-humble-ament-lint-common/jammy,now 0.12.11-1jammy.20240728.201429 amd64 [installed,automatic]
ros-humble-ament-lint/jammy,now 0.12.11-1jammy.20240728.191833 amd64 [installed,automatic]
ros-humble-ament-package/jammy,now 0.14.0-4jammy.20220421.093117 amd64 [installed,automatic]
ros-humble-ament-pep257/jammy,now 0.12.11-1jammy.20240728.194154 amd64 [installed,automatic]
ros-humble-ament-uncrustify/jammy,now 0.12.11-1jammy.20240728.195917 amd64 [installed,automatic]
ros-humble-ament-xmllint/jammy,now 0.12.11-1jammy.20240728.200027 amd64 [installed,automatic]
ros-humble-angles/jammy,now 1.15.0-1jammy.20240728.193007 amd64 [installed]
ros-humble-backward-ros/jammy,now 1.0.5-1jammy.20240728.203139 amd64 [installed,automatic]
ros-humble-bicycle-steering-controller/jammy,now 2.37.2-1jammy.20240823.151640 amd64 [installed,automatic]
ros-humble-builtin-interfaces/jammy,now 1.2.1-1jammy.20240728.211104 amd64 [installed,automatic]
ros-humble-class-loader/jammy,now 2.2.0-3jammy.20240728.205127 amd64 [installed,automatic]
ros-humble-common-interfaces/jammy,now 4.2.4-1jammy.20240728.222111 amd64 [installed,automatic]
ros-humble-composition-interfaces/jammy,now 1.2.1-1jammy.20240728.211951 amd64 [installed,automatic]
[... 517 lines cut ...]
[move_group-3] [INFO] [1726509613.470232504] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-3] [INFO] [1726509613.471924812] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-3] [INFO] [1726509613.474293660] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[move_group-3] [INFO] [1726509613.475880584] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[move_group-3] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[move_group-3]          at line 127 in ./src/class_loader.cpp
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 46001]
[INFO] [franka_gripper_node-7]: process has finished cleanly [pid 46021]
[INFO] [joint_state_publisher-5]: process has finished cleanly [pid 46007]
[ERROR] [rviz2-1]: process has died [pid 45999, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/temp/franka_ros2_ws/install/franka_fr3_moveit_config/share/franka_fr3_moveit_config/rviz/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_7fdxny70 --params-file /tmp/launch_params_p7zbg3an --params-file /tmp/launch_params_y2mh_wz8 --params-file /tmp/launch_params_l6wosrbx'].
[ERROR] [move_group-3]: process has died [pid 46003, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_hy8y7hg3 --params-file /tmp/launch_params_sm_ev5kf --params-file /tmp/launch_params_8f3cdmmj --params-file /tmp/launch_params_aiagjn78 --params-file /tmp/launch_params_2xbbcabn --params-file /tmp/launch_params_rqwj9awi --params-file /tmp/launch_params_xhlfi8k0'].
[move_group-3] 
[ERROR] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process[ros2 run controller_manager spawner joint_state_broadcaster-9] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[... cut ...]
[ERROR] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process has died [pid 46027, exit code -15, cmd 'ros2 run controller_manager spawner joint_state_broadcaster'].
[ERROR] [ros2 run controller_manager spawner fr3_arm_controller-8]: process has died [pid 46023, exit code -15, cmd 'ros2 run controller_manager spawner fr3_arm_controller'].
```

</details>

Has anyone seen this error? I am new to ROS and our FR3 robot so any advice in the right direction is greatly appreciated. I am hesitant to simply increase all the velocity values at the risk of damaging our robot.

The robot hasn't moved since its purchase, so I think it's in its home position.

adamrankin, 2024-09-16:
Minor update: I have enabled realtime scheduling and that warning no longer appears

adamrankin, 2024-09-20:
Ok, with libfranka 0.14.1 I now get
`[FrankaHardwareInterface]: libfranka: Incompatible library version (server version: 7, library version: 8). Please check https://frankaemika.github.io for Panda system updates or choose a libfranka version that uses the server version 7 from the table at https://frankaemika.github.io/docs/compatibility.html`

but my robot has no updates available. Is server version 8 some sort of beta version for the FR3?

AndreasKuhner, 2024-09-26:
Hey @adamrankin ,
indeed the server version 8 is still some internal release... meaning the the 0.13.x is the one for 5.6.0. On the other side, 0.14.x will be for our next 5.7.0 release.

If you get a joint_velocity_violation, moveit tries to execute something with too high velocities. Maybe you can try to reduce the maximum velocity and try again? 

Cheers,
Andreas

adamrankin, 2024-09-27:
Thank you so much for the help! I will revert to 0.13.x and proceed from there.

With regards to moveit, I'm just trying to launch the first example `ros2 launch franka_fr3_moveit_config moveit.launch.py ...`. My basic read through doesn't show any move commands sent when launching rvis.

1) Are there any move commands sent when launching this configuration (maybe the robot homes? or some test movement?)
2) (sorry, I'm new to ROS) What configuration values can I change to allow a slightly higher velocity? Or are the values provided by this repo the maximum save limit for the FR3?

Thank you again for your help,
Adam

AndreasKuhner, 2024-09-30:
Hi @adamrankin ,
I guess to see if it generally works... you can try to run some of the other example controllers to check if the setup works in general. (e.g. gravity compensation or joint impedance example). 

Cheers,
Andreas

AndreasKuhner, 2024-10-25:
If you have set any limits within Watchmen (aka the safety), maybe try to remove them. 

Do you maybe have a gripper? And did you configure it?
Maybe you can also try to guide the robot into a neutral position far away from joint limits and start the examples there.

adamrankin, 2024-10-28:
`Maybe you can also try to guide the robot into a neutral position far away from joint limits and start the examples there.`

This was it. I moved the robot out of pack position and tried it, and it worked. I will put it back to pack position and see if it occurs again.

closed 2024-10-28 by adamrankin

adamrankin, 2024-10-29:
I can confirm that after moving back to pack pose, this error re-occurs. It's an easy work-around to just move the robot a bit, but maybe this is worthy of a bug?

AndreasKuhner, 2024-10-30:
Technically seen it is not a bug. While being in the pack-pose, the robot is close/within the joint limits which means for safety 'please don't move very fast'. Hence, you first need to get out of the packing pose for normal operation :smile: 

It is not a bug.. it is a feature :grin: