frankarobotics/franka_ros2 #73: franka_description package version mismatch

Roboticshwx, 2024-10-24:
My environment is
Ubuntu 22.04;
ROS 2 Humble;
libfranka v0.14.2;
franka_ros2 v0.1.15

After I use colcon build to construct the ros2 package, I found the command:
ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true can't run
because of a inertia properties errors;

Last, I found my franka_description package version is incorrect, it shows 0.2.0 instead of 0.3.0; It should be 0.3.0 to compatibilize above version.

I solve the problem by removing the package and use git clone to clone the newest version (0.3.0), and it succeed.

Can anyone tell me why this issue happen? I am curious about that~

Thanks a lot~

AndreasKuhner, 2024-10-25:
Hi @Roboticshwx ,
in the 'old' franka_description, we had inertia values which worked but didn't fulfill all expected metrics - so we updated them with the version 0.3.0 to also be recognized as proper inertias.

Cheers,
Andreas

Roboticshwx, 2024-10-25:
> Hi @Roboticshwx , in the 'old' franka_description, we had inertia values which worked but didn't fulfill all expected metrics - so we updated them with the version 0.3.0 to also be recognized as proper inertias.
> 
> Cheers, Andreas

Thanks for replying. 
Can it be seen as a bug of `franka_ros2` project.? 

Because I tried several time, after `colcon build` of the franka_ros2. It construct a v0.2.0 `franka_description`, which is not compatible.

Hope your team can fix this problem.

Thanks,
hwx

AndreasKuhner, 2024-10-25:
> after colcon build of the franka_ros2. It construct a v0.2.0

Theoretically, all should be fine. Do you have the correct franka_ros2 version checked out as well as the correct franka_description version? Maybe also a clean of the install/build folder sometimes helps - colcon sometimes doesn't clean up older stuff.

closed 2025-03-05 by AndreasKuhner