frankarobotics/franka_ros2 #75: Cannot launch gazebo bringup

JuanCarlosHR2003, 2024-11-01:
I have been trying to use the Gazebo launcher, but the Franka robot cannot be visualized because Gazebo cannot load the .dae files from the franka_description package.  All other packages work fine. 
[image]

Gazebo version: 11.10.2

JuanCarlosHR2003, 2024-11-01:
I managed to visualize the Franka arm changing the utils.xacro. I changed the filename of the visual and collision meshes to "file://$(find franka_description)/meshes..."
However, when I try to visualize the robot end effector, I have the same problem. Even with the patch I used with the Franka arm, the problem persists.
Here is a photo of the log.
[image]

As well, an important detail to add is that I can visualize and control the robot end effector in RVIZ with no problems.

AndreasKuhner, 2024-11-04:
Hi @JuanCarlosHR2003 ,
what exactly did you change from what string to what other string? 

Maybe you can also give us the version you are using + franka_ros2 version so that we could check. At least we didn't see any problems at the moment :smile: 

Cheers,
Andreas

TianxiaoYee, 2024-12-17:
I have same problem. I am using libfranka 0.14.1 and ros2 humble

PyoSH, 2024-12-28:
I also had same problem. (ubuntu 22.04 + ros2 humble + libfranka 0.14.1)
As @JuanCarlosHR2003 said, I changed file name in franka_description/robots/common/utils.xacro 
 
origin) mesh filename="package://${description_pkg}/meshes/robot_arms/${arm_id}/visual/${name}.dae" 
replace) mesh filename="file://$(find franka_description)/meshes/robot_arms/${arm_id}/visual/${name}.dae"

[image]

After with these replacement, do colcon build.
[image]

AndreasKuhner, 2025-01-07:
Thanks @PyoSH for the input. I think we have a similar fix on its way... but not yet released.

Cheers,
Andreas

AndreasKuhner, 2025-03-05:
Should be done. If not, let us know, thanks.

closed 2025-03-05 by AndreasKuhner