frankarobotics/franka_ros2 #76: New Joint Velocity Error On Example Effort Controller.

glvov-bdai, 2024-11-12:
So, the examples previously worked on system version 5.6, and libfranka version 0.13, but on some of our arms, we have system version 5.7.1 and libfranka version 0.14.0.

On the arms with the newer system software/libfranka, when we run the gravity compensation example controller, we run into a joint velocity error that we previously did not experience on the older software version. We can successfully run the example controllers which command joint velocities such as the joint velocity example controller. It appears that any example controller that commands efforts produces a joint velocity limit error immediately as the arm starts up. 

This is preventing us from operating our arms on the new firmware version (5.7). We have re-set the mass properties to the defaults, but the issue still persists. Are there any other troubleshooting steps we can take for this issue? 

```
glvov@BDAI:/workspaces$ ros2 launch franka_bringup gravity_compensation_example_controller.launch.py robot_ip:=<OUR_IP>
[INFO] [launch]: All log files can be found below /home/glvov/.ros/log/2024-11-12-22-28-19-510121-P3-WS02-67120
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher-1]: process started with pid [67121]
[INFO] [robot_state_publisher-2]: process started with pid [67123]
[INFO] [ros2_control_node-3]: process started with pid [67125]
[INFO] [spawner-4]: process started with pid [67127]
[INFO] [spawner-5]: process started with pid [67129]
[INFO] [franka_gripper_node-6]: process started with pid [67131]
[INFO] [spawner-7]: process started with pid [67133]
[franka_gripper_node-6] [INFO] [1731450499.924335518] [fr3_gripper]: Trying to establish a connection with the gripper
[franka_gripper_node-6] [INFO] [1731450499.925185459] [fr3_gripper]: Connected to gripper
[ros2_control_node-3] [WARN] [1731450499.944939571] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[robot_state_publisher-2] [WARN] [1731450499.945051052] [kdl_parser]: The root link fr3_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] [INFO] [1731450499.945110707] [robot_state_publisher]: got segment fr3_hand
[robot_state_publisher-2] [INFO] [1731450499.945132656] [robot_state_publisher]: got segment fr3_hand_tcp
[robot_state_publisher-2] [INFO] [1731450499.945136655] [robot_state_publisher]: got segment fr3_leftfinger
[robot_state_publisher-2] [INFO] [1731450499.945139546] [robot_state_publisher]: got segment fr3_link0
[robot_state_publisher-2] [INFO] [1731450499.945142387] [robot_state_publisher]: got segment fr3_link1
[robot_state_publisher-2] [INFO] [1731450499.945145168] [robot_state_publisher]: got segment fr3_link2
[robot_state_publisher-2] [INFO] [1731450499.945147858] [robot_state_publisher]: got segment fr3_link3
[robot_state_publisher-2] [INFO] [1731450499.945150697] [robot_state_publisher]: got segment fr3_link4
[robot_state_publisher-2] [INFO] [1731450499.945153480] [robot_state_publisher]: got segment fr3_link5
[robot_state_publisher-2] [INFO] [1731450499.945156206] [robot_state_publisher]: got segment fr3_link6
[robot_state_publisher-2] [INFO] [1731450499.945158992] [robot_state_publisher]: got segment fr3_link7
[robot_state_publisher-2] [INFO] [1731450499.945161634] [robot_state_publisher]: got segment fr3_link8
[robot_state_publisher-2] [INFO] [1731450499.945164274] [robot_state_publisher]: got segment fr3_rightfinger
[ros2_control_node-3] [INFO] [1731450499.945144234] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1731450499.951567316] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1731450499.951627835] [FrankaHardwareInterface]: Connecting to robot at "10.103.2.109" ...
[ros2_control_node-3] [INFO] [1731450499.959794707] [FrankaHardwareInterface]: Successfully connected to robot
[ros2_control_node-3] [INFO] [1731450499.962527595] [service_server]: Service started
[ros2_control_node-3] [INFO] [1731450500.066416818] [action_server]: Action server started
[ros2_control_node-3] [INFO] [1731450500.066486343] [resource_manager]: Successful initialization of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1731450500.066651901] [resource_manager]: 'configure' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1731450500.066669786] [resource_manager]: Successful 'configure' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1731450500.066685614] [resource_manager]: 'activate' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1731450500.067884474] [FrankaHardwareInterface]: Started
[ros2_control_node-3] [INFO] [1731450500.067901969] [resource_manager]: Successful 'activate' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1731450500.070049416] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [INFO] [1731450500.071813697] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[joint_state_publisher-1] [INFO] [1731450500.218822661] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[ros2_control_node-3] [INFO] [1731450500.230308722] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-4] [INFO] [1731450500.247318774] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1731450500.247964333] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1731450500.248058412] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-4] [INFO] [1731450500.252504314] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[ros2_control_node-3] [INFO] [1731450500.258843356] [controller_manager]: Loading controller 'franka_robot_state_broadcaster'
[ros2_control_node-3] [INFO] [1731450500.264707912] [controller_manager]: Loading controller 'gravity_compensation_example_controller'
[spawner-5] [INFO] [1731450500.280532937] [spawner_franka_robot_state_broadcaster]: Loaded franka_robot_state_broadcaster
[spawner-7] [INFO] [1731450500.280666064] [spawner_gravity_compensation_example_controller]: Loaded gravity_compensation_example_controller
[ros2_control_node-3] [INFO] [1731450500.281156900] [controller_manager]: Configuring controller 'franka_robot_state_broadcaster'
[ros2_control_node-3] [INFO] [1731450500.301187283] [controller_manager]: Configuring controller 'gravity_compensation_example_controller'
[spawner-5] [INFO] [1731450500.304450348] [spawner_franka_robot_state_broadcaster]: Configured and activated franka_robot_state_broadcaster
[INFO] [spawner-4]: process has finished cleanly [pid 67127]
[INFO] [spawner-5]: process has finished cleanly [pid 67129]
[ros2_control_node-3] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-3]   what():  libfranka: Move command aborted: motion aborted by reflex! ["joint_velocity_violation"]
[ERROR] [ros2_control_node-3]: process has died [pid 67125, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /workspaces/bdai/_build/install/franka_bringup/share/franka_bringup/config/controllers.yaml --params-file /tmp/launch_params_bndpm7pl --params-file /tmp/launch_params_tn9dp3vs --params-file /tmp/launch_params_q6gqi342 -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-3] was required: shutting down launched system
[INFO] [spawner-7]: sending signal 'SIGINT' to process[spawner-7]
[INFO] [franka_gripper_node-6]: sending signal 'SIGINT' to process[franka_gripper_node-6]
[INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
[... 134 lines cut ...]
[spawner-5] [INFO] [1731451149.018510863] [spawner_franka_robot_state_broadcaster]: Configured and activated franka_robot_state_broadcaster
[ros2_control_node-3] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-3]   what():  libfranka: Move command aborted: motion aborted by reflex! ["joint_velocity_violation"]
[... cut ...]
[INFO] [franka_gripper_node-6]: process has finished cleanly [pid 67627]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 67619]
[ERROR] [spawner-7]: process[spawner-7] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [spawner-7]: sending signal 'SIGTERM' to process[spawner-7]
[ERROR] [spawner-7]: process has died [pid 67629, exit code -15, cmd '/opt/ros/humble/lib/controller_manager/spawner gravity_compensation_example_controller --ros-args'].
glvov@BDAI:/workspaces/$ 

```
@glvov-bdai

glvov-bdai, 2024-11-12:
Resolved by power cycling the arm

closed 2024-11-12 by glvov-bdai

AndreasKuhner, 2024-11-13:
Hi @glvov-bdai ,
if you use libfranka version 0.14.0, then I can recommend to use 0.14.2. Especially, if you use libfranka with ROS 2, it has some juicy fixes on board :smile: 

Cheers,
Andreas