frankarobotics/franka_ros2 #77: moveit_py and franka_ros2

adamrankin, 2024-11-15:
I have a situation where I need to be able to use moveit_py and franka_ros2, but moveit_py isn't available for humble, and franka_ros2 doesn't compile for iron.

In general, would you recommend that I try to compile moveit_py for humble, or try to debug franka_ros2 for iron (or perhaps another solution I haven't thought of)?

I've tried the approach here:
https://moveit.ai/moveit%202/ros/2023/05/31/balancing-stability-and-development.html
but that led to libsrdfdom version incompatibility issues

AndreasKuhner, 2024-11-18:
Hi @adamrankin ,
are you sure that moveit_py isn't available for humble? Maybe I don't understand this page https://index.ros.org/p/moveit_py/#humble but it should be available.. no?

Otherwise, I also read in some thread to compile moveit for humble from source - at least the `moveit_py` bindings should then also come along.

Cheers,
Andreas

adamrankin, 2024-11-18:
Yes, moving the moveit repo leads to libsrdfdom version conflicts. Moveit uses 2.0.5 but something in humble still uses 2.0.4. I will investigate.

adamrankin, 2024-11-21:
After much exploration what I'm really looking for is a python interface to MoveGroupInterface, which doesn't exist for ros2. moveit_py is more about creating your own motion planning instance, whereas I want to send messages to an existing motion planning instance.

This is barely related to the above question, but moveit_py for humble was what started me on this path.

For anyone who reads this, I just made a symbolic link for 2.0.4 to 2.0.5 to avoid the issue.

closed 2024-11-21 by adamrankin

AndreasKuhner, 2024-11-22:
Thank you for sharing :smile: