frankarobotics/franka_ros2 #79: Can I use franka_ros2 with Ubuntu 22.04 ?

shortcastovercoffee, 2024-11-21:
Can I use franka_ros2 with Ubuntu 22.04 without having to get the premium subscription for Ubuntu?

shortcastovercoffee, 2024-11-21:
Did you mean 20.04? because when I read the documentation it says "For Ubuntu 22.04, we recommend using the [Ubuntu Pro real-time kernel](https://ubuntu.com/real-time). After enabling it, you can skip directly to [Allow a user to set real-time permissions for its processes](https://frankaemika.github.io/docs/installation_linux.html#installation-real-time)."

AndreasKuhner, 2024-11-22:
I meant the 22.04 - there you can do everything without needing a premium subscription :smile:

yijionglin, 2025-01-21:
@AndreasKuhner Hi Andreas,

Would you recommend using ROS 1 or ROS 2 for controlling the Franka robot? Or do you think ros2_franka is now as well-developed as ros_franka?

All the best,
Bourne

AndreasKuhner, 2025-01-22:
4.0.0 means that you guys are using a Panda robot..?

yijionglin, 2025-01-22:
Yes…

AndreasKuhner, 2025-01-23:
I see. If you use a Panda, we don't officially support it with ROS 2. For Panda we only have official support for ROS 1.

We officially support ROS 2 with our FR3.

yijionglin, 2025-01-23:
Thanks for the clarification! 

Just wondering where I could check the version update features note so that I could learn what can or can't be used with the 4.0.0 version.

Many thanks,
Bourne

AndreasKuhner, 2025-01-23:
Here you go: https://frankaemika.github.io/docs/compatibility.html 

(look for the robot version **FER** which is the Panda while FR3 is... well FR3 😄 )

closed 2025-03-05 by AndreasKuhner