frankarobotics/franka_ros2 #80: Failed   <<< franka_hardware

Vojislav032, 2024-11-26:
I have a problem with installation of franka_ros2 on ubuntu 22.04, ros2 humble. 
Problem is it want install franka_hardware.
error log is here:[Error log](https://github.com/user-attachments/files/17925794/Error.log)
Can you help me to complete installation?

pala5, 2024-11-27:
Same problem here. 
I was trying to figure out which is the problem, because few weeks ago I can colcon build it without any issue, but this morning appears to me the same error detailed by @Vojislav032.
I am working on ubuntu 22.04.
I am building the libfranka (0.14.1), following both the [libfranka README](https://github.com/frankaemika/libfranka?tab=readme-ov-file#clone-the-repository) and the [franka_ros2/.devcontainer/Dockerfile](https://github.com/frankaemika/franka_ros2/blob/humble/.devcontainer/Dockerfile), without success.
Analyzing the error, it seems that the current available version of pinocchio from robotpkg public repository was updated 3 weeks ago to the 3.3.0

Vojislav032, 2024-11-27:
I walk trough installation again from: 
1. https://stack-of-tasks.github.io/pinocchio/download.html
2. https://github.com/frankaemika/libfranka?tab=readme-ov-file#clone-the-repository
3. https://github.com/frankaemika/franka_ros2
And installation runs smoothly: [done log](https://github.com/user-attachments/files/17939964/done.log)
But test setup:
ros2 launch franka_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true
fails...

AndreasKuhner, 2024-12-02:
Hi @Vojislav032 ,
what error do you get if you run the test? And did it succeed before?

Cheers,
Andreas

Vojislav032, 2024-12-02:
Hi @AndreasKuhner ,
Test error log file is here:
[Test Error log](https://github.com/user-attachments/files/17983623/Test.Error.log)
I am using this system first time.

AndreasKuhner, 2024-12-03:
Hi @Vojislav032 ,
there is a small typo in our docu... the package you would use for that example is `franka_fr3_moveit_config` not `franka_moveit_config` - does this work?

Vojislav032, 2024-12-03:
Hi @AndreasKuhner 
it doesn't work. 
[Test Error log2.txt](https://github.com/user-attachments/files/17989900/Test.Error.log2.txt)
problem is with installation of franka_ros2...
installation fails after 5/14 parts installed log is here....
[install fail log.txt](https://github.com/user-attachments/files/17989981/install.fail.log.txt)
i don`t know how to install it...
i have try to install it from vsCode and i have a problem to.
[image]
Do you maybe have VM with ros2, franaka_ros2, moveit?

AndreasKuhner, 2024-12-04:
The docker container usually works. 

But your normal installation as well -> did you **source** your **setup.sh** are doing the **colcon build**? Aka, being in the top level folder where you see the `install` folder doing `source install/setup.sh`.

Vojislav032, 2024-12-04:
i think i have made it work. the problem was with Ubuntu Pro (is was not enabled) when i power it on and   made source install/setup.sh. it works properly.

AndreasKuhner, 2024-12-10:
Good to hear :smile: 

If you have further questions, just shoot them ahead. If not yet, maybe we can close the issue.

Cheers,
Andreas

closed 2024-12-10 by Vojislav032