frankarobotics/franka_ros2 #84: I would like to ask if it is possible to obtain the model parameters of the robot in the simulation environment, such as M, C, G

JiangShangJiu, 2024-12-10:
I would like to ask if it is possible to obtain the model parameters of the robot in the simulation environment, such as M, C, G

AndreasKuhner, 2024-12-10:
Hi @JiangShangJiu ,
not directly while using e.g. gazebo (but maybe gazebo provides them...). Then again, you could look into https://github.com/stack-of-tasks/pinocchio. You have access to our URDF which Pinocchio uses for computing all kinds of dynamic parameters (like M, C, G) and we also use it nowadays in libfranka for computing those values: https://github.com/frankaemika/libfranka/blob/main/src/robot_model.cpp.

Cheers,
Andreas

JiangShangJiu, 2024-12-10:
Thank you so much for your quick answer, I used franka_ros simulation in the ROS1 notetic version, I remember that I could get the dynamics parameters of the robot, as well as the Jacobian matrix. However, when I use the ROS2 Humble version to use franka_ros2 to emulate in the ignition_gazebo, I don't find these interfaces, is it no longer available?

JiangShangJiu, 2024-12-10:
@AndreasKuhner Thank you so much for your quick answer, I used franka_ros simulation in the ROS1 notetic version, I remember that I could get the dynamics parameters of the robot, as well as the Jacobian matrix. However, when I use the ROS2 Humble version to use franka_ros2 to emulate in the ignition_gazebo, I don't find these interfaces, is it no longer available?

AndreasKuhner, 2024-12-10:
This is not yet available - and probably has a low enough priority to never come directly :sweat_smile: 

Main reason is that we thought using Pinocchio is nowadays so easy, that we don't need to expose yet another interface which is exactly doing this. If you are using our franka_ros2/libfranka, you already have pinocchio on your system. So checking out how we do it in libfranka gives you already a blueprint on how to do it - copy/paste 4tw :smile:

AndreasKuhner, 2025-03-05:
If there are still some open questions, let us know :)