frankarobotics/franka_ros2 #87: Support for ROS 2 Jazzy?

pbaist, 2024-12-17:
Hi,

Assuming I've understood things correctly, franka_ros2 only supports ROS 2 Humble (and indeed fails to compile under Jazzy following the instructions as they are.)

As ROS 2 Jazzy is over half a year old at this point, and the de facto LTS version of ROS 2 for the next few years, do you have an estimate for when it will be supported? Additionally, makeshift/temporary instructions to get franka_ros2 working under Jazzy right now would be appreciated.

Thanks,

AndreasKuhner, 2024-12-18:
Hi @pbaist ,
yes, so far we only support `humble`. Yet, getting it to compile under `Jazzy` is already on our roadmap :partying_face: 

Cheers,
Andreas

JuanSeSand, 2025-01-14:
Hi @AndreasKuhner, do you also plan to support a Galactic version? or, is there someone who has adapted the existing packages to be used under Galactic?

AndreasKuhner, 2025-01-15:
Hi @JuanSeSand ,
no, we don't plan to have a support for Galactic. The first support long-term version is humble and soonish Jazzy.

But why do you want to use this old version?

Cheers,
Andreas

lianghongzhuo, 2025-05-15:
Any updates regarding this? :)

lianghongzhuo, 2025-05-15:
With some minimal effort, I made a working (compiling version) of franka_ros2 in jazzy.
https://github.com/frankaemika/franka_ros2/compare/humble...lianghongzhuo:franka_ros2:jazzy
This ignores some components, but it compiles the essential component of this repo: the franka hardware.
Maybe it is a starting point?

AndreasKuhner, 2025-06-04:
Hi @tonybaltovski ,
unfortunately no. It sits in our backlog but other priorities always push it back again 😒 

Cheers,
Andreas

adamrankin, 2025-06-04:
@lianghongzhuo I recreated your changes and was able to compile, but I was only partially successful when running the rviz2 sample when connected to our robot. I will continue to investigate

lusu8892, 2025-06-05:
Please add support to Jazzy. ROS2 Humble ros control do not have `is_async` for different hardware. This could lead to thread blocking issue if I ran franka with an Robotiq gripper for instance. While Jazzy has this `is_async` flag that could make other hardware running on its own thread and its own frequency.

kmohyeldine, 2025-06-05:
Hi @lusu8892 @tonybaltovski! sorry for the delay we are a bit busy! i will prioritize it soon and let you know!

AndreasKuhner, 2025-08-06:
The curious eye might have already found this one..: https://github.com/frankarobotics/franka_ros2/pull/170. 

Still in PR phase and testing phase - so no guarantee that everything works, but if you wanna give it a test right -> Go for it. Feedback in the PR is also welcome (e.g. bugs).

Cheers,
Andreas