frankarobotics/franka_ros2 #91: Some problems

jin-yiwei, 2024-12-24:
Hello~
Environment: ubuntu22.04 + libfranka0.14.1 + franka_ros2 + realtime kernel

Problem: 
1. libfranka example is ok except "force_control", the error is: "libfranka: Move command aborted: motion aborted by reflex! ["controller_torque_discontinuity"]"
2. all franka_ros2 example is not ok, the error in 2 different PC is: 

- [ros2_control_node-3] terminate called after throwing an instance of 'St9bad_alloc'; [ros2_control_node-3]   what():  std::bad_alloc

- reflex

Do you know why? thanks!

AndreasKuhner, 2025-03-05:
If you still have some open questions, please let us know. Cheers :)

closed 2025-03-05 by AndreasKuhner

jin-yiwei, 2025-04-01:
Thank you for your reply. I now feel that my question is more likely to be the first one. I have developed a joint position example, attempting to achieve position control for any given joint, combined with https://github.com/frankaemika/franka_ros2/blob/humble/franka_example_controllers/src/motion_generator.cpp I have written a motion controller, and I think I should have followed the restrictions on speed, acceleration, and jerk mentioned in FCI, but still get the error “what(): libfranka: Move command aborted: motion aborted by reflex! ["joint_motion_generator_velocity_discontinuity", "joint_motion_generator_acceleration_discontinuity"]”， May I ask why this is?