frankarobotics/franka_ros2 #92: Realtime control of Franka Hands

prakharg01, 2025-01-08:
### **Delay in Real-Time Control of Franka Grippers**

### Description:  
We are encountering significant delays when controlling the Franka grippers in ROS2 and librabka using the `move` or `grasp` modes. The delay between triggering a gripper action and observing its execution is too high for our use case, which requires real-time control. 

### Request:
Is there any way to reduce the delay or achieve real-time control of the gripper? Any guidance or recommendations for optimization would be greatly appreciated.

Thank you for your support!

ignacio-pm, 2025-03-06:
Hi @prakharg01  ,

thank you for your feedback. We are aware of the limitations of the Franka Hand for real-time control and we will work in a new interface to achieve a better real-time performance. For the moment, there is no possibility to achieve a real-time control of the gripper.

Best,
Ignacio

cwoodhayes, 2026-08-20:
I see this and #100 have both been closed; I was not able to find any update in the repo regarding this issue. Is enabling real-time control of the hand no longer on the roadmap? @AndreasKuhner 
If it's en route, I'd really love to use it for an upcoming paper; otherwise, we will have to order a different gripper :(