frankarobotics/franka_ros2 #93: Not able to connect to the fr3 using ros2

Yash-Kashiv, 2025-01-15:
My environment 
Ubuntu 22.04;
ROS 2 Humble;
libfranka v0.14.2;
franka_ros2 v0.1.15
franka_description v0.4.0

when using the command 
ros2 launch franka_bringup franka.launch.py arm_id:=fr3 robot_ip:=192.168.1.11 use_rviz:=true load_gripper:=false

this is the error i am getting 
[ros2_control_node-3] [INFO] [1736965827.996626890] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1736965827.996703608] [FrankaHardwareInterface]: Connecting to robot at "192.168.1.11" ...
[ros2_control_node-3] [FATAL] [1736965827.999922100] [FrankaHardwareInterface]: Could not connect to robot
[ros2_control_node-3] [FATAL] [1736965827.999950581] [FrankaHardwareInterface]: libfranka: Unexpected response while handling Get Robot Model command!
[ros2_control_node-3] [INFO] [1736965827.999971453] [resource_manager]: Failed to initialize hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [WARN] [1736965828.000041468] [resource_manager]: System hardware component 'FrankaHardwareInterface' from plugin 'franka_hardware/FrankaHardwareInterface' failed to initialize.
[ros2_control_node-3] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-3]   what():  Wrong state or command interface configuration.
[ros2_control_node-3] missing state interfaces:
[ros2_control_node-3] ' fr3_joint1/position '	' fr3_joint1/velocity '	' fr3_joint1/effort '	' fr3_joint2/position '	' fr3_joint2/velocity '	' fr3_joint2/effort '	' fr3_joint3/position '	' fr3_joint3/velocity '	' fr3_joint3/effort '	' fr3_joint4/position '	' fr3_joint4/velocity '	' fr3_joint4/effort '	' fr3_joint5/position '	' fr3_joint5/velocity '	' fr3_joint5/effort '	' fr3_joint6/position '	' fr3_joint6/velocity '	' fr3_joint6/effort '	' fr3_joint7/position '	' fr3_joint7/velocity '	' fr3_joint7/effort '	
[ros2_control_node-3] missing command interfaces:
[ros2_control_node-3] ' fr3_joint1/position '	' fr3_joint1/velocity '	' fr3_joint1/effort '	' fr3_joint2/position '	' fr3_joint2/velocity '	' fr3_joint2/effort '	' fr3_joint3/position '	' fr3_joint3/velocity '	' fr3_joint3/effort '	' fr3_joint4/position '	' fr3_joint4/velocity '	' fr3_joint4/effort '	' fr3_joint5/position '	' fr3_joint5/velocity '	' fr3_joint5/effort '	' fr3_joint6/position '	' fr3_joint6/velocity '	' fr3_joint6/effort '	' fr3_joint7/position '	' fr3_joint7/velocity '	' fr3_joint7/effort '	
[ros2_control_node-3] Stack trace (most recent call last):

AndreasKuhner, 2025-01-16:
Hi @Yash-Kashiv ,
what image version do you use? Is it 5.7.0 or 5.7.1?

Cheers,
Andreas

Yash-Kashiv, 2025-01-17:
Hi @AndreasKuhner,
Could you please elaborate on which image version you are referring to? I am new to robotics and want to make sure I understand correctly.

AndreasKuhner, 2025-01-17:
If you go to Desk (the interface you get to via your browser), you can go to `settings` and in the `dashboard` tab you will find the image version of the robot.

Yash-Kashiv, 2025-01-17:
@AndreasKuhner thanks for the description. Image version is 5.7.1

AndreasKuhner, 2025-01-17:
Thanks... looks good regarding version matching.

Do you get this error all the time? If the network connection was broken, this might also occur.

AndreasKuhner, 2025-03-05:
If you still have issues, please let us know. Thanks 😄

closed 2025-03-05 by AndreasKuhner