frankarobotics/franka_ros2 #95: Colcon build fatal error

aleada23, 2025-01-23:
Hi,
while following the steps in order to install the code the colcon build fails and the following error is showed. I tried several ways to fix it, but nothing worked. Thank you in advance.

> Starting >>> franka_msgs
Starting >>> integration_launch_testing
Finished <<< integration_launch_testing [2.05s]                  
Finished <<< franka_msgs [23.0s]                        
Starting >>> franka_hardware
Starting >>> franka_gripper
--- stderr: franka_gripper                                                
In file included from /home/yumi/franka_ros2_ws/src/franka_gripper/include/franka_gripper/gripper_action_server.hpp:28,
                 from /home/yumi/franka_ros2_ws/src/franka_gripper/src/gripper_action_server.cpp:31:
/home/yumi/franka_ros2_ws/build/franka_msgs/rosidl_generator_cpp/franka_msgs/action/grasp.hpp:10:10: fatal error: franka_msgs/action/detail/grasp__type_support.hpp: No such file or directory
   10 | #include "franka_msgs/action/detail/grasp__type_support.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/gripper_server.dir/build.make:76: CMakeFiles/gripper_server.dir/src/gripper_action_server.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/gripper_server.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
>Failed   <<< franka_gripper [7.38s, exited with code 2]
Aborted  <<< franka_hardware [9.60s]                                 

>Summary: 2 packages finished [32.8s]
  1 package failed: franka_gripper
  1 package aborted: franka_hardware
  2 packages had stderr output: franka_gripper franka_hardware
  8 packages not processed

Smith-JackSmith, 2025-01-23:
Can you please ensure you building on the tag 0.1.15

We recommend using one of the container build methods.

Smith-JackSmith, 2025-01-23:
We built a generic container based upon Ubuntu 22.04.

Then followed the [local machine installation](https://github.com/frankaemika/franka_ros2?tab=readme-ov-file#local-machine-installation) instructions, and built successfully.

We are unable to reproduce your results.

Can you please provide details about your setup ?