frankarobotics/franka_ros2 #97: Colcon error while building franka_ign_ros2_control

pvskand, 2025-01-25:
Hi there,

I've been struggling with the following build error:

```
/usr/include/ignition/msgs8/ignition/msgs/world_modify.pb.h: In member function ‘void ignition::msgs::WorldModify::unsafe_arena_set_allocated_cloned_uri(std::string*)’:
/usr/include/ignition/msgs8/ignition/msgs/world_modify.pb.h:618:15: error: ‘struct google::protobuf::internal::ArenaStringPtr’ has no member named ‘UnsafeArenaSetAllocated’
  618 |   cloned_uri_.UnsafeArenaSetAllocated(&::PROTOBUF_NAMESPACE_ID::internal::GetEmptyStringAlreadyInited(),
      |               ^~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/franka_ign_ros2_control-system.dir/build.make:76: CMakeFiles/franka_ign_ros2_control-system.dir/src/ign_ros2_control_plugin.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/franka_ign_ros2_control-system.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< franka_ign_ros2_control [16.0s, exited with code 2]
Aborted  <<< franka_robot_state_broadcaster [14.4s]
Aborted  <<< franka_example_controllers [14.6s]
```
I get the same error when I building with or without the `-DFranka_DIR` cmake flag. 

Here's my system information:

Ubuntu `22.04`
ROS 2 Humble
Robot system version: `5.6.0`
libfranka `v0.13.3`
`protoc --version : libprotoc 3.12.4`

I'm building `franka_ros2` from the `v0.1.15` tag by downloading the `.tar.gz` file from [here](https://github.com/frankaemika/franka_ros2/releases/tag/v0.1.15).

I'm not sure what the issue is -- and I suspect it has to do something with the `protobuf` version.  Any help on this front would be great!

AndreasKuhner, 2025-01-27:
Hi @pvskand ,
what build environment do you use? Usually, our `franka_ros2` docker environment has all the dependencies it needs. 

So, cloning our `franka_ros2` from github, open it with visual code which should ask you for setting up its docker environment and then the usual `colcon build` - that should work at the moment. Could you try that?

(The Readme would also give you a direct guidance on how to do it)

pvskand, 2025-01-28:
Thanks for the response. Installing it via docker was smooth! My only worry is that docker would add a layer of latency but I guess for the time being, it is manageable.

closed 2025-01-28 by pvskand

AndreasKuhner, 2025-01-28:
Nowadays that isn't a problem anymore - if you run into timing issues, the problem usually lies somewhere else :)