frankarobotics/franka_ros2 #101: Error while building franka_ros2

Cody-Vu, 2025-02-04:
Hello everyone,

I installed libfranka (ver. 0.13.2) and after I want to install franka_ros2 I stuck at the building process `colcon` build --symlink-install --cmake-args `-DCMAKE_BUILD_TYPE=Release.` I receive this error:

/home//franka_ros2_ws/src/franka_hardware/src/franka_hardware_interface.cpp:21:10: fatal error: franka/logging/logger.hpp: No such file or directory
   21 | #include <franka/logging/logger.hpp>

 This is the output from `ls` `/usr/local/include/franka/`:

active_control_base.h  active_motion_generator.h  control_tools.h  duration.h  exception.h  gripper_state.h  lowpass_filter.h  rate_limiting.h  robot_state.h     vacuum_gripper_state.h
active_control.h       active_torque_control.h    control_types.h  errors.h    gripper.h    log.h            model.h           robot.h          vacuum_gripper.h

I guess there exist no file named logger.hpp but what am I supposed to do now?
I'd appreciate any help!

AndreasKuhner, 2025-02-05:
Hi @Cody-Vu ,
I guess you picked up the latest `franka_ros2` humble branch which comes with some dependencies towards libfranka...

Depending on what robot version you use (you see that in Desk/Settings), you need to use different `libfranka/franka_ros2` versions.

Here you can see a picture of which version needs what: https://frankaemika.github.io/docs/compatibility.html

If you want to use latest `franka_ros2` then you also need to use the latest `libfranka` (but this only works for image version 5.7.2!)

Cheers,
Andreas

Cody-Vu, 2025-02-05:
> Hi [@Cody-Vu](https://github.com/Cody-Vu) , I guess you picked up the latest `franka_ros2` humble branch which comes with some dependencies towards libfranka...
>
> Depending on what robot version you [...]

Hi @AndreasKuhner ,

I see thanks for clarification. I just realized that the robot I'm using (Panda FER) is not compatible with franka_ros2. I've seen some community projects that makes franka_ros2 compatible with older libfranka versions. But would you recommend me to stick with ROS1(franka_ros) for the panda? I'm just asking to explore all options to choose which setup will be the best.
Thanks in advance!

Kind regards,
Cody

AndreasKuhner, 2025-02-05:
Okey, this explains a lot xd.

We know that franka_ros works stable for the old Panda robot. The franka_ros2 adaptions are coming from the community. So, I can't give you an official recommendation for that 😁 But it seems to be used...

Cheers,
Andreas