frankarobotics/franka_ros2 #102: error build humble branch

lianghongzhuo, 2025-02-06:
fatal error: franka_robot_state_broadcaster/franka_robot_state_broadcaster_parameters.hpp: No such file or directory
in the include folder there is only one file:
ls franka_ros2/franka_robot_state_broadcaster/include/franka_robot_state_broadcaster
get only:
franka_robot_state_broadcaster.hpp

lianghongzhuo, 2025-02-06:
soldved, turns out it's a generated header, but cmake did not find it correctly

closed 2025-02-06 by lianghongzhuo

sudohaider, 2025-03-04:
> soldved, turns out it's a generated header, but cmake did not find it correctly

@lianghongzhuo I am facing the same issue. Could you tell how you resolved it?

AndreasKuhner, 2025-03-04:
Hi @sudohaider ,
sometimes it helps to make a `rm -rf build/ install/` to get rid of old depdendencies and do a `colcon build` again.

Cheers,
Andreas

lianghongzhuo, 2025-03-04:
> > soldved, turns out it's a generated header, but cmake did not find it correctly
> 
> [@lianghongzhuo](https://github.com/lianghongzhuo) I am facing the same issue. Could you tell how you resolved it?

this header file is auto generated, I manually put this header file in my source include folder
see here: https://github.com/lianghongzhuo/franka_ros2/commit/554a9581236861d38da95ef1528e649f10d4d752