frankarobotics/franka_ros2 #103: Issue: Incompatible libfranka Version After System Update

VivekSai07, 2025-02-06:
I recently updated the Franka system from version `5.7.1` to `5.7.2`, and then installed the latest version of `libfranka`. However, I am facing issues with launching the Franka robot in `franka_ros2`.

### Installation Steps:
1. **Updated the Franka system** from version `5.7.1` to `5.7.2`.
2. **Installed the latest version of libfranka** – version `0.15.0`:
   - [libfranka GitHub README](https://github.com/frankaemika/libfranka/blob/main/README.md)
3. **Installed franka_ros2** by following the guide:
   - [franka_ros2 GitHub README](https://github.com/frankaemika/franka_ros2/blob/humble/README.md)

## Error Description

When trying to launch the `moveit.launch.py` file using the command:

```bash
ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=192.168.1.11
```

The following errors occur:
```
[INFO] [launch]: All log files can be found below /home/lwu/.ros/log/2025-01-30-15-54-35-118312-Dell3680-82609
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [82612]
[INFO] [robot_state_publisher-2]: process started with pid [82614]
[INFO] [move_group-3]: process started with pid [82616]
[INFO] [ros2_control_node-4]: process started with pid [82618]
[INFO] [joint_state_publisher-5]: process started with pid [82620]
[INFO] [spawner-6]: process started with pid [82622]
[INFO] [franka_gripper_node-7]: process started with pid [82624]
[INFO] [ros2 run controller_manager spawner fr3_arm_controller-8]: process started with pid [82626]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process started with pid [82628]
[ros2_control_node-4] [WARN] [1738248875.364191908] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[franka_gripper_node-7] [INFO] [1738248875.364215420] [fr3_gripper]: Trying to establish a connection with the gripper
[ros2_control_node-4] text not specified in the prefix tag
[ros2_control_node-4] text not specified in the arm_prefix tag
[ros2_control_node-4] [INFO] [1738248875.364385629] [resource_manager]: Loading hardware 'FrankaHardwareInterface'
[franka_gripper_node-7] [INFO] [1738248875.364786874] [fr3_gripper]: Connected to gripper
[ros2_control_node-4] [INFO] [1738248875.366441306] [resource_manager]: Initialize hardware 'FrankaHardwareInterface'
[ros2_control_node-4] [INFO] [1738248875.366473191] [FrankaHardwareInterface]: Connecting to robot at "192.168.1.11" ...
[ros2_control_node-4] [FATAL] [1738248875.367425475] [FrankaHardwareInterface]: Could not connect to robot
[ros2_control_node-4] [FATAL] [1738248875.367430735] [FrankaHardwareInterface]: libfranka: Incompatible library version (server version: 9, library version: 7). Please check https://frankaemika.github.io for Panda system updates or choose a libfranka version that uses the server version 9 from the table at https://frankaemika.github.io/docs/compatibility.html.
[ros2_control_node-4] [INFO] [1738248875.367437345] [resource_manager]: Failed to initialize hardware 'FrankaHardwareInterface'
```

## Library Version Check

I checked the version of the libfranka library using the following command:
```
dpkg -l | grep libfranka
```
The output is as follows:
```
ii  ros-humble-libfranka                               0.13.6-1jammy.20241202.160951           amd64        libfranka is a C++ library for Franka Robotics research robots
```
## Troubleshooting Steps
1. I deleted the existing logs of libfranka library.
2. I tried rebuilding the workspace using colcon:
```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
This resulted in the following error during the build process:
```
Starting >>> franka_msgs
Starting >>> integration_launch_testing
Finished <<< integration_launch_testing [0.54s]
Finished <<< franka_msgs [8.04s]
Starting >>> franka_hardware
Starting >>> franka_gripper
--- stderr: franka_gripper
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "FindFranka.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Franka", but
  CMake did not find one.

  Could not find a package configuration file provided by "Franka" (requested version 0.13.2) with any of the following names:

    FrankaConfig.cmake
    franka-config.cmake

  Add the installation prefix of "Franka" to CMAKE_PREFIX_PATH or set "Franka_DIR" to a directory containing one of the above files. If "Franka" provides a separate development package or SDK, be sure it has been installed.

---
Failed   <<< franka_gripper [0.73s, exited with code 1]
Aborted  <<< franka_hardware [0.74s]

Summary: 2 packages finished [8.88s]
  1 package failed: franka_gripper
  1 package aborted: franka_hardware
  2 packages had stderr output: franka_gripper franka_hardware
  8 packages not processed
```
## CMakeLists.txt Investigation
After opening the CMakeLists.txt files from the franka_gripper, franka_hardware, and franka_semantic_components packages, I noticed references to the Franka package which could be causing the issue. The error mentioned the inability to find FrankaConfig.cmake or franka-config.cmake.

[image]

[image]

[image]

AndreasKuhner, 2025-02-07:
Hi @VivekSai07 ,
the key point of your logs is the `Server version`: `libfranka: Incompatible library version (server version: 9, library version: 7).`. Meaning, you `franka_ros2` is still using the old `libfranka-common` 0.13.x version -> yes, `libfranka-common` not `libfranka`. The common library is a submodule... so, by any change -> when you checked out the libfranka repo did you also do a `git submodule update`? 

Anyways, you need to remove `ros-humble-libfranka` first -> this version is outdated (it still is 0.13.2). Then you can build libfranka, cpack it and install 🥳 

Cheers,
Andreas

VivekSai07, 2025-02-07:
> Hi [@VivekSai07](https://github.com/VivekSai07) , the key point of your logs is the `Server version`: `libfranka: Incompatible library version (server version: 9, library version: 7).`. Meaning, you [...]

Thank you @AndreasKuhner for your reply. But the franka_ros2 by default installs ros-humble-libfranka 0.13.2. I've installed a global version of libfranka (0.15.0), but it'll not be used by franka_ros2, even if I remove ros-humble-libfranka 0.13.2.

I tried rebuilding the workspace using colcon:
```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
This resulted in the following error during the build process:
```
Starting >>> franka_msgs
Starting >>> integration_launch_testing
Finished <<< integration_launch_testing [0.54s]
Finished <<< franka_msgs [8.04s]
Starting >>> franka_hardware
Starting >>> franka_gripper
--- stderr: franka_gripper
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "FindFranka.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Franka", but
  CMake did not find one.

  Could not find a package configuration file provided by "Franka" (requested version 0.13.2) with any of the following names:

    FrankaConfig.cmake
    franka-config.cmake

  Add the installation prefix of "Franka" to CMAKE_PREFIX_PATH or set "Franka_DIR" to a directory containing one of the above files. If "Franka" provides a separate development package or SDK, be sure it has been installed.

---
Failed   <<< franka_gripper [0.73s, exited with code 1]
Aborted  <<< franka_hardware [0.74s]

Summary: 2 packages finished [8.88s]
  1 package failed: franka_gripper
  1 package aborted: franka_hardware
  2 packages had stderr output: franka_gripper franka_hardware
  8 packages not processed
```

AndreasKuhner, 2025-02-12:
Meaning, you build `libfranka`, did a `cpack` and an `sudo dpkg -i libfranka-.....deb`?

VivekSai07, 2025-02-12:
> Meaning, you build `libfranka`, did a `cpack` and an `sudo dpkg -i libfranka-.....deb`?

Yes, I've followed the installation guide provided in the [libfranka repository](https://github.com/frankaemika/libfranka/blob/main/README.md).

AndreasKuhner, 2025-02-12:
Can you then check if under `/usr/lib/` a `libfranka.so` is?

If yes, you might try to remove the 'version' dependency in the `CMakeLists.txt` of `franka_hardware` within the `franka_ros2` folder. There, you should find a `find_package(Franka 0.13.2 REQUIRED)` -> you can try to remove the `0.13.2` and just have `find_package(Franka REQUIRED)`.

I hope that helps. Although I am a bit confused why it doesn't complain with my setup 🤔 

Cheers,
Andreas

DimiSf, 2025-02-13:
Hello ,
I face the same issue now, i followed your advice at my issue https://github.com/frankaemika/libfranka/issues/176 .
I now can run the echo example **but :**
```
dpkg -l | grep libfranka 
ii  ros-humble-libfranka                               0.13.6-1jammy.20241202.160951            amd64        libfranka is a C++ library f
```
even though i followed building from source from github repo for system version 5.7.2 and for libfranka 0.15 
i get : at libfranka dir CMakeList.txt 

[image]

Same , when building franka_ros2 from github repo :
```
~/franka_ros2_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> franka_msgs
Starting >>> integration_launch_testing
Finished <<< integration_launch_testing [1.09s]                  
Finished <<< franka_msgs [20.5s]                       
Starting >>> franka_hardware
Starting >>> franka_gripper
--- stderr: franka_hardware            
/home/group/franka_ros2_ws/src/franka_hardware/src/franka_hardware_interface.cpp:21:10: fatal error: franka/logging/logger.hpp: No such file or directory
   21 | #include <franka/logging/logger.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/franka_hardware.dir/build.make:104: CMakeFiles/franka_hardware.dir/src/franka_hardware_interface.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/group/franka_ros2_ws/src/franka_hardware/src/ros_libfranka_logger.cpp:1:
/home/group/franka_ros2_ws/src/franka_hardware/include/franka_hardware/ros_libfranka_logger.hpp:5:10: fatal error: franka/logging/logging_sink_interface.hpp: No such file or directory
    5 | #include <franka/logging/logging_sink_interface.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/franka_hardware.dir/build.make:146: CMakeFiles/franka_hardware.dir/src/ros_libfranka_logger.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:195: CMakeFiles/franka_hardware.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< franka_hardware [11.0s, exited with code 2]
Aborted  <<< franka_gripper [14.4s]   

Summary: 2 packages finished [35.1s]
  1 package failed: franka_hardware
  1 package aborted: franka_gripper
  1 package had stderr output: franka_hardware
  8 packages not processed
```
I checked /usr/lib/ about libfranka.so it isnt there but 

```
find /usr /usr/local /opt -name "libfranka.so*" 2>/dev/null
/opt/ros/humble/lib/x86_64-linux-gnu/libfranka.so
/opt/ros/humble/lib/x86_64-linux-gnu/libfranka.so.0.13.6
/opt/ros/humble/lib/x86_64-linux-gnu/libfranka.so.0.13
```
it is quite strange since i followed the instructions from here: https://github.com/frankaemika/libfranka
what should i consider and do ?
Thank you,
Dimitra.

AndreasKuhner, 2025-02-14:
Hi @DimiSf ,
can you try to un-install `ros-humble-libfranka`? It looks like that it still picks up the wrong version.

Then building with libfranka 0.15.0 and franka_ros2 1.0.0 should (hopefully) work.

Cheers,
Andreas

mrceki, 2025-02-28:
Hi, I also had the same issue with version 0.15.0; my system version is 5.6.0. So I uninstalled `ros-humble-libfranka` and built it from the source, but that also didn't work. I downgraded `libfranka` to 0.13.3 from source and now it works. 

My setup was ok until `ros-humble-libfranka`  upgraded.