frankarobotics/franka_ros2 #105: Failed to lock the realtime publisher after 10 attempts

filipul1s, 2025-02-12:
Hi, everyone
I use the latest version of franka_ros2, but the following problems occur. 
[franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
May I ask what is the reason?

AndreasKuhner, 2025-02-12:
Hi @filipul1s ,
a recent ROS 2 update put this issue a bit more prominent on display...

Basically, each cycle our `robot_state_broadcaster` is trying to get access to the data. Yet, it can be that during asking for access, already another agent (aka the one writing the data), still uses the shared resource... therefore the `robot_state_broadcaster` is getting a reject and throws an error - not necessarily the behavior we want.

We are already on a fix for that which will soon be on main 🥳 

Cheers,
Andreas

AndreasKuhner, 2025-02-12:
Hi @ssrpo ,
the easiest would be, if you just pull the latest `main` which has the fix. Of course, that state depends on OS `5.7.2` + libfranka 0.15.0. Current main is more or less franka_ros2 '1.0.1'.

Would that work 😄 ?

Cheers,
Andreas

Smith-JackSmith, 2025-02-13:
@filipul1s

> [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts

Can you please indicate how frequently you are seeing this message ?

filipul1s, 2025-02-13:
> [@filipul1s](https://github.com/filipul1s)
> 
> > [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
> 
> Can you please indicate how frequently you are seeing this message ?

When I run the joint_position_example_controller example, it prints this error until the program interrupts.

Smith-JackSmith, 2025-02-13:
> [@Smith-JackSmith](https://github.com/Smith-JackSmith)
> 
> > [@filipul1s](https://github.com/filipul1s)
> > > [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
> > 
> > 
> > Can you please indicate how frequently you are seeing this message ?
> 
> When I run the joint_position_example_controller example, it prints this error until the program interrupts.

Yes, but how frequently ?

ssrpo, 2025-02-13:
> Hi [@ssrpo](https://github.com/ssrpo) , the easiest would be, if you just pull the latest `main` which has the fix. Of course, that state depends on OS `5.7.2` + libfranka 0.15.0. Current main is more or less franka_ros2 '1.0.1'.
> 
> Would that work 😄 ?
> 
> Cheers, Andreas

Hi @AndreasKuhner, 

Thanks a lot for the quick reply. 
Yeah it seems like the best idea. 

However I would prefer to update the OS once you have a stable new tag for franka_ros2. 
It is my understanding that once I update the robot to 5.7.2 I can't go back to 5.7.1, right? (other people at the lab are using the robot and it wouldn't be nice to break things for too long ^^ ). 

Do you have a timeline for the new tag? :) 

@Smith-JackSmith On my case I was getting the error at each cycle (pretty red terminal while launching any controller). I'll have a robot to test tomorrow I can save logs or smth else if that can help you. Let me know. 

------------------------------------------------------------------------------------------------------------------------------
Some logs I copied yesterday if useful (the moveit error was due to a bad locale config, working fine now on dummy hardware):

[ros2_control_node-4] [ERROR] [1739357357.731889463] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'

[rviz2-1] [INFO] [1739358005.145276971] [moveit_robot_model.robot_model]: Loading robot model 'fr3'...

[rviz2-1] [WARN] [1739358005.145280015] [moveit_robot_model.robot_model]: Skipping virtual joint 'virtual_joint' because its child frame 'fr3_link0' does not match the URDF frame 'base'

[rviz2-1] [INFO] [1739358005.145287944] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint

[rviz2-1] [ERROR] [1739358005.147584408] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'fr3_arm.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {fr3_arm.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.

[ros2_control_node-4] [ERROR] [1739358010.918387761] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster' 
----------------------------------------------------------------------------------------------------------------------------

Feliz dia, 

Susana

jcarpinelli-bdai, 2025-02-13:
> ...if you just pull the latest `main` which has the fix. Of course, that state depends on OS `5.7.2` + libfranka 0.15.0. 

By `main`, do you mean `humble`? There is no `main` branch in this (public) repository. 

I am using `libfranka-v0.15.0` and fimwate `5.7.2` and I'm seeing this error on the realtime patch for Ubuntu very frequently. Within a minute, the screen is filled with `ERROR` logging statements. 

We actually found this issue documented by #105 when we upgraded `libfranka` and firmware versions to address https://github.com/frankaemika/franka_ros2/issues/94#issuecomment-2634314151. #94 is a similar issue where the `franka_robot_state_broadcaster` is returning `ERROR` frequently.

Is it possible to remove all `return_type::ERROR` returns unless it's truly a critical error, and instead log warnings once per minute or so?

bjin-bdai, 2025-02-13:
> Is it possible to remove all return_type::ERROR returns unless it's truly a critical error, and instead log warnings once per minute or so?

Another possibility maybe is making that threshold configurable, so certain users can increase the failure threshold if they're willing to accept staler state information

AndreasKuhner, 2025-02-14:
> By main, do you mean humble? There is no main branch in this (public) repository.

yes, sorry. My head is somehow programmed with calling everything main which is the default branch 😄 

For a quick local fix, you can definitely just change the error to a success. As you say, the data might be then sometimes a bit stale, then again... 1khz should still be having lots of new data. 

We also pushed a fix to the `humble` branch. Does the error still occur so regularly for you guys? In the fix, we try a few times to get access to the data... which resolved the issue mostly. But timings are different for different machines.. so, could still be a problem.

@kmohyeldin Shall we make another round of fix?

jcarpinelli-bdai, 2025-02-14:
>  Does the error still occur so regularly for you guys? In the fix, we try a few times to get access to the data... which resolved the issue mostly. 

We are still seeing the issue. I see in the source code you're trying 10 times, but we still receive many errors printed to the log and screen. I think another fix would be helpful. If possible, it would be great if the fix did not require upgrading `libfranka` or Franka firmware. Having to upgrade a full lab of robots with firmware while also updating software is costly!

Smith-JackSmith, 2025-02-17:
> I have the same issue, terminal keep printing: [ros2_control_node-4] [ERROR] [1739784681.723445214] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
> 
> but most programs still work, I'm currently using libfranka 0.14.2, firmware 5.7.1 and franka_ros 0.1.15. This error polluted terminal by pouring this info about every 1.5s. I hope this message could help for debugging.

Can you please clarify whether you are using **_franka_ros_** or **franka_ros2**.
Thank You.

Trinitro-Dopamine, 2025-02-17:
> Can you please clarify whether you are using **_franka_ros_** or **franka_ros2**. Thank You.

it's franka_ros2, I updated my last comment.

Trinitro-Dopamine, 2025-02-17:
could this issue result the failure to IK algorithm? Amoung those examples, joint_impedance_with_ik_example_controller couldn't run properly and even jam the arm, flashing red and warn a emergency stop saying "The error has been triggered by an inconsistency in the sensor reading of the safe input." However it's fine while I use RVIZ for motion planning, I could submit the terminal log if you need extra information.

filipul1s, 2025-02-18:
> Yes, but how frequently ?

This is the output from running move_to_start_example_controller.launch.py, timestamped as follows:
[ros2_control_node-3] [ERROR] [1739845044.368623992] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845044.368671523] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845044.369580382] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845044.369604564] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845044.370608248] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845044.370634074] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845044.371592699] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845044.371614385] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845044.372673881] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845044.372696060] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845047.369682353] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845047.369712977] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845051.388553325] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845051.388598969] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845051.389582775] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845051.389604787] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845058.413890127] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845058.413922739] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845059.400312169] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845059.400341612] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845059.401290476] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845059.401304485] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-3] [ERROR] [1739845059.402282614] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts
[ros2_control_node-3] [ERROR] [1739845059.402295375] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'

Smith-JackSmith, 2025-02-18:
Thank you , @filipul1s for the log.
I can also reproduce this type of behavior, but, only by "loading" my host with extra tasks.
When the host is quiet, I only rarely see these messages.

There is some evidence that defining NON_POLLING at compile time will improve this issue - although it may not completely disappear (in fact almost definitely will not).

You can do something like this within  franka_robot_state_broadcaster.hpp (line 25)
franka_robot_state_broadcaster/include/franka_robot_state_broadcaster/franka_robot_state_broadcaster.hpp

```
// define NON_POLLING before including realtime_publisher to ensure that
// the realtime_tools::RealtimePublisher uses a better synchronization strategy than sleeping for 500 Microseconds !
#define NON_POLLING TRUE
#include <realtime_tools/realtime_publisher.hpp>
```

I will confer with my colleagues about getting this to be the default.

Smith-JackSmith, 2025-03-17:
A recent commit should reduce the frequency of this error message - which I should mention, lies deep within the ROS2 RealTimePublisher code, used by franka_robot_state_broadcaster (so, it's really non-trivial to work around).

Be aware, this change will probably not eliminate the error, but may help reduce the number of messages.  We will be porting the entire franka_ros2 project to ROS2 Jazzy soon, which may improve this issue, since the real fix to this issue resulted in a compete re-write of several low level ROS2 components; the fix was only partially back-ported to humble.

That being said, it is important to realize that external factors may be in play here -  underspec'd  / overloaded hardware (the machine running this code) is likely to be an issue.

cheng9911, 2025-03-20:
I have pulled the latest code from the repository. When I run the command ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=172.16.0.2, the program operates normally for about 40 minutes before it crashes, producing the following error messages:
`[ros2_control_node-4] [ERROR] [1742453826.852454204] [franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts 

[ros2_control_node-4] [ERROR] [1742453826.852504640] [controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'
[ros2_control_node-4] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-4]   what():  libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
[ros2_control_node-4] Stack trace (most recent call last) in thread 66958:
[ros2_control_node-4] #18   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-4] #17   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f27c0b8a84f]
[ros2_control_node-4] #16   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f27c0af8ac2]
[ros2_control_node-4] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f27c0d89252, in 
[ros2_control_node-4] #14   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x557de0d1608d, in 
[ros2_control_node-4] #13   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f27c11d6334, in controller_manager::ControllerManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f27c0963bac, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #11   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f27c09813ef, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #10   Object "/home/sun/Documents/Franka/build/franka_hardware/libfranka_hardware.so", at 0x7f27b99ae941, in franka_hardware::FrankaHardwareInterface::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #9    Object "/home/sun/Documents/Franka/build/franka_hardware/libfranka_hardware.so", at 0x7f27b99dae62, in franka_hardware::Robot::readOnce()
[ros2_control_node-4] #8    Object "/home/sun/Documents/Franka/install/libfranka/lib/libfranka.so.0.15.0", at 0x7f27b98a5de7, in franka::ActiveControl::readOnce()
[ros2_control_node-4] #7    Object "/home/sun/Documents/Franka/install/libfranka/lib/libfranka.so.0.15.0", at 0x7f27b98a38a2, in franka::Robot::Impl::throwOnMotionError(franka::RobotState const&, unsigned int) [clone .cold]
[ros2_control_node-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f27c0d5b4d7, in __cxa_throw
[ros2_control_node-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f27c0d5b276, in std::terminate()
[ros2_control_node-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f27c0d5b20b, in 
[ros2_control_node-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f27c0d4fb9d, in 
[ros2_control_node-4] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f27c0a8c7f2]
[ros2_control_node-4] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f27c0aa6475]
[ros2_control_node-4] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-4]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-4]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f27c0afa9fc]
[ros2_control_node-4] Aborted (Signal sent by tkill() 66831 1000)
[ERROR] [ros2_control_node-4]: process has died [pid 66831, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_lwsck8sz --params-file /home/sun/Documents/Franka/install/franka_fr3_moveit_config/share/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-4] was required: shutting down launched system
[INFO] [franka_gripper_node-7]: sending signal 'SIGINT' to process[franka_gripper_node-7]
[INFO] [joint_state_publisher-5]: sending signal 'SIGINT' to process[joint_state_publisher-5]
[INFO] [move_group-3]: sending signal 'SIGINT' to process[move_group-3]
[INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
[INFO] [rviz2-1]: sending signal 'SIGINT' to process[rviz2-1]
[franka_gripper_node-7] [INFO] [1742454250.862981810] [rclcpp]: signal_handler(signum=2)
[move_group-3] [INFO] [1742454250.864013011] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1742454250.864828428] [rclcpp]: signal_handler(signum=2)
[rviz2-1] [INFO] [1742454250.865563902] [rclcpp]: signal_handler(signum=2)
[move_group-3] [INFO] [1742454250.874302157] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-3] [INFO] [1742454250.874893240] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-3] [INFO] [1742454250.875445992] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[move_group-3] [INFO] [1742454250.875661204] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[... cut ...]
[move_group-3] 
[ERROR] [rviz2-1]: process has died [pid 66825, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/sun/Documents/Franka/install/franka_fr3_moveit_config/share/franka_fr3_moveit_config/rviz/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_wj57wozy --params-file /tmp/launch_params_nkfcaq9r --params-file /tmp/launch_params_h2ai1n68 --params-file /tmp/launch_params_cyqecjy9'].
[INFO] [franka_gripper_node-7]: process has finished cleanly [pid 66837]
`

falfab, 2025-10-01:
@AndreasKuhner, are there any updates on this? 
We are currently facing this issue that prevents us from using any of the newly purchased robots.

We are running on this setup:

OS: Ubuntu 22.04 (PREEMPT_RT)
ROS: Humble
librfranka: 0.15.0
franka_ros2: humble