frankarobotics/franka_ros2 #106: Failed to build in Docker container

moss-coleman, 2025-02-14:
Hi,

I'm getting the error below when trying to build franka_ros2 with Docker using the instructions for "Option A: using Docker Compose". The container builds but it is failing when building the ROS packages in the container. I've tried to edit the Docker file to build the latest libfranka, tag 0.15.0, from source but that approach is getting other issues. 

Thank you in advance for any help with this, and please let me know if you need any other information about my setup.

/ros2_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> franka_msgs
Starting >>> integration_launch_testing
Finished <<< integration_launch_testing [0.83s]                  
Finished <<< franka_msgs [14.8s]                        
Starting >>> franka_hardware
Starting >>> franka_gripper
--- stderr: franka_hardware                                               
/ros2_ws/src/franka_hardware/src/franka_hardware_interface.cpp:21:10: fatal error: franka/logging/logger.hpp: No such file or directory
   21 | #include <franka/logging/logger.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/franka_hardware.dir/build.make:104: CMakeFiles/franka_hardware.dir/src/franka_hardware_interface.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /ros2_ws/src/franka_hardware/src/ros_libfranka_logger.cpp:15:
/ros2_ws/src/franka_hardware/include/franka_hardware/ros_libfranka_logger.hpp:19:10: fatal error: franka/logging/logging_sink_interface.hpp: No such file or directory
   19 | #include <franka/logging/logging_sink_interface.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/franka_hardware.dir/build.make:146: CMakeFiles/franka_hardware.dir/src/ros_libfranka_logger.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:195: CMakeFiles/franka_hardware.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< franka_hardware [6.39s, exited with code 2]
Aborted  <<< franka_gripper [11.6s]                                   

Summary: 2 packages finished [26.6s]
  1 package failed: franka_hardware
  1 package aborted: franka_gripper
  1 package had stderr output: franka_hardware
  8 packages not processed

francescosemeraro, 2025-02-14:
Go for the instructions with VSCode Container, they are easier to follow and debug

moss-coleman, 2025-02-14:
Thanks for the advice Francesco. Unfortunately, the same error as above happens with the VSCode .devcontainer. The container builds in both cases, the issue is building franka_ros2 after the container is built with colcon.

Smith-JackSmith, 2025-02-16:
Can you indicate which branch or tag you are trying to build.

moss-coleman, 2025-02-17:
Hi Jack, I was using the 'humble' branch set to the latest tag 'v1.0.0'. I noticed today that the latest commits weren't being pulled down using this. With these latest commits, everything is working. Thanks for the response anyway :)

closed 2025-02-17 by moss-coleman