frankarobotics/franka_ros2 #108: communication_constraints_violation

byeongkyu, 2025-02-18:
```
...

[ros2_control_node-6] [ERROR] [1739844272.283399633] [franka.franka_robot_state_broadcaster]: Failed to lock the realtime publisher after 10 attempts                                                                                                     
[ros2_control_node-6] [ERROR] [1739844272.283458191] [franka.controller_manager]: The update call of the following controller returned an error: 'franka_robot_state_broadcaster'                                                                         
[ros2_control_node-6] terminate called after throwing an instance of 'franka::ControlException'                                                                                                                                                           
[ros2_control_node-6]   what():  libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]                                                                                                                       
[ros2_control_node-6] Stack trace (most recent call last) in thread 1767:                                                                                                                                                                                 
[ros2_control_node-6] #18   Object "", at 0xffffffffffffffff, in              

...
```

Kernel: 6.8.2-rt11 #1 SMP PREEMPT_RT
FCI: 5.7.2
libfranka: 0.15.0
ROS2: humble

Self ping test: 
10000 packets transmitted, 10000 received, 0% packet loss, time 10089ms
rtt min/avg/max/mdev = 0.131/0.209/0.452/0.041 ms

I am currently operating an FR3 robotic arm using ROS2. During operation, the communication_constraints_violation error occurs frequently. I would like to identify the cause and find a solution.

filipul1s, 2025-02-18:
I have the same problem.

byeongkyu, 2025-02-18:
It is difficult to understand why a single exception causes all launches to terminate. After reviewing the source code, I found that a franka::ControlException occurs while executing robot_->readOnce(). This code is located on line 173 of franka_hardware_interface.cpp. As a temporary measure, I modified this part as follows, and the connection remains stable without any issues.

```
try {
    hw_franka_robot_state_ = robot_->readOnce();
  }
  catch(franka::ControlException &e) {
    RCLCPP_WARN(getLogger(), "Exceptions: %s", e.what());
    return hardware_interface::return_type::OK;
  }
```

AndreasKuhner, 2025-02-18:
Hi @byeongkyu ,
the problem is that the backend is going into an error mode. So, I don't know how the system is controllable if you do this.

Having communication violations is often caused by some missed details - you might check https://frankaemika.github.io/docs/troubleshooting.html to give some ideas what to check :)

Cheers,
Andreas

byeongkyu, 2025-02-19:
Hi @AndreasKuhner 

I have tried all the methods from the link you provided, and they all worked. As I initially noted, almost all test results show no errors. What other approaches could there be?

On a different note, I find it hard to understand why a single error should cause all operations to stop. It would be great if there were a way to recover quickly even when a communication violation error occurs.

jcarpinelli-bdai, 2025-02-26:
We have also seen this issue on multiple arms. The issue started occurring after we upgraded firmware from 5.6 to 5.7.2, and after we upgraded to `libfranka-v0.15.0` and `franka_ros2-v1.0.0`. We have the arms directly connected to each computer. We see this issue consistently after 15 minutes (or so) of the `ros2_control_node` starting. The `communication_test` shows 100% passing.

AndreasKuhner, 2025-02-28:
@jcarpinelli-bdai Could you maybe change the `communication_test` to not stop after 10 seconds but run e.g. for 20 minutes to see, if you also get there then the communication violation?

shr-eyas, 2025-03-14:
Facing the same issue, but only when update has some computation going on in it.

AndreasKuhner, 2025-03-31:
What do you mean by 'our Franka ROS driver'? Do you mean an adapted variant to handle multiple robots on one machine?

In the single robot case, you didn't run into a communication violation?

obiza-theaiinstitute, 2025-03-31:
> What do you mean by 'our Franka ROS driver'? Do you mean an adapted variant to handle multiple robots on one machine?
We are using the open-source Franka ROS2 driver; “humble” on franka_ros2.
We run it twice with different namespaces on a single real-time machine to control both arms.

> In the single robot case, you didn't run into a communication violation?
Yes, single-arm stations also run into it, usually after 30 minutes.

`communication_constraints_violation` happens frequently with firmware 5.7.2 (libfranka 0.15.0). Downgrading to firmware 5.6.0 (libfranka 0.13.3) fixes it.

AndreasKuhner, 2025-03-31:
Could you download the logs via Desk and share it with us?

joestg, 2025-03-31:
This issue still persist. The RAI Institute met with Franka team (Dmitrij Forssilow, Jeremie Tardivon, Kamal Mohy El Dine) on 3/25 to discuss bugs.  
During that meeting, RAI decided to wait until FW 5.8 before further debugging this issue. We choose to roll back to FW 5.6.0 until then, since the bug is not present in 5.6.0. 

I may have wiped out the logs by installing FW 5.6.0 and doing a factory reset.

ruoshiwen, 2025-04-03:
We met the same problem for two week after a firmware upgrade to 5.7.2. Following your advice @obiza-theaiinstitute , we downgraded to firmware 5.6.0 (libfranka 0.13.3, franka_ros2 0.15.0), now it seems running fine. Many thanks!

mrceki, 2025-04-04:
Hi all, 
I had the same issue. I tried everything except downgrading the libfranka version. Then I downgraded libfranka to 0.13.3 from 0.13.6, and it's running fine. Thanks!

Jelmerdw, 2025-06-18:
@obiza-theaiinstitute @ruoshiwen @mrceki 

We are trying to downgrade as well. However, we cannot find this option in Franka Desk or Franka World. Could you elaborate how you downgrade the robot?

kmohyeldine, 2025-06-18:
@Jelmerdw  please contact support@franka.de

zadaianchuk, 2025-07-02:
We encountered a similar problem with libfranka 0.15 and Franka firmware 5.8, so it seems like downgrading is the only option so far

Jelmerdw, 2025-08-11:
@AndreasKuhner 

We tested the release of system image 5.8.1 and this seems to work much better. With this system image, we are also able to use the robot in combination with Jazzy:

https://github.com/frankarobotics/franka_ros2/pull/170

We now encounter the communication_constraints_violation only when we try to move too soon after launching the robot or when we relaunch directly after a previous launch.

lusu8892, 2025-08-12:
On my end, 5.8.1 robot system did not improve stability (still get communication_constraints_violation)
My setup:
PC running real-time kernal, high performance
ROS2: Humble
franka_ros2: v1.0.0
franka_description: 0.4.0
libfranka: 0.15.0

Communication_test: 100% passed

Tests have ran with ROS2 and franka_ros2:
`franka_fr3_moveit_config moveit.launch.py`, just launch this without sending any trajectory, run 10-15mins then got the error message.
`franka_bringup joint_impedance_example_controller.launch.py`, run 10-15 mins dead.

**Solely running controller example from libfranka does not cause communication_violation at all.......**
Modified joint_impedance_controller running 30mins, no error got.

This may not be conclusive, but look like either Franka_ros2 or ROS2 is not stable........

Please continue working on this issue..

Thanks!

danielsanjosepro, 2025-08-14:
Hey! I was going to give the release 5.8.1 a try but it does not look like it is available when I try to update my system...

AndreasKuhner, 2025-08-26:
Hi @danielsanjosepro ,
which version are you currently running? 

Cheers,
Andreas

bjin-bdai, 2025-11-04:
> With ROS 2, we already know that is unfortunately not aligned with a 'real time' control scheme. ros2_control is doing a great job but the problem here is: It runs with its own spinning time while libfranka syncs up with the spin of the robot. Meaning, with libfranka, after doing the `read`, you get the 'full time' to do your computations and send the next command back. Yet, with ROS 2 you have always a 'sleep' after the 'write' (controlled by ros2_control). This sleep is used to aim for the target frequency. However, that means that your 'read' can get delayed and you don't get the 'full time' to do your computations.

@AndreasKuhner this is interesting, so is the implication here that ideally the `ros2_control_node.cpp` loop would just be something like:

```
while (running) {
    cm->read(...);
    cm->update(...);
    cm->write(...);

    // don't sleep because read() will synchronize the loop
}
```

For context, we've run into communication constrain violations with firmware 5.8, but with a `humble` branch, even though `wireshark` data does not show noticeable packet delays between our Linux box and Franka control box

bjin-bdai, 2025-11-04:
> Our repo franka_ros2 jazzy branch currently have a fork of [controller_manager](https://github.com/frankarobotics/franka_ros2/tree/jazzy/controller_manager) which should have the necessary fix for the Jazzy distro to work with Franka robots

@BarisYazici would it be possible to have this implementation ported to the `humble` branch as well?

lianghongzhuo, 2025-11-05:
> Our repo franka_ros2 jazzy branch currently have a fork of [controller_manager](https://github.com/frankarobotics/franka_ros2/tree/jazzy/controller_manager) which should have the necessary fix for the Jazzy distro to work with Franka robots

this fix seems not solved the problem on my side, I am using 5.8.2 firmware with ros jazzy. Still get "communication_constraints_violation" error every 30 minutes (in average, some times even quicker)

BarisYazici, 2025-11-05:
To have the same baseline can all of you test the communication with franka_ros2 `gravity_compensation_example` on the 5.8.2 image on Humble or Jazzy distros without another custom controller involved?

AndreasKuhner, 2025-11-06:
> @AndreasKuhner this is interesting, so is the implication here that ideally the ros2_control_node.cpp loop would just be something like:

@bjin-bdai You hit the spot :) Our libfranka/FCI really puts the user in the middle of the closed control loop. If we now have something, which runs on its own schedule (aka sleeps on its own), there will be an unavoidable drift over time. I think with humble, it can run for a longer time but with our initial push to jazzy, the problem was thrown a lot more in our face. So yes: Ideally, libfranka can do the 'sleep' (will do it anyways) and ros2_control doesn't block or control frequencies.

> this fix seems not solved the problem on my side, I am using 5.8.2 firmware with ros jazzy. Still get "communication_constraints_violation" error every 30 minutes (in average, some times even quicker)

Next week, we will have the image version 5.9.0 release which should contain more juicy updates to ease the issue about `communication_constraint_violations`. Technically, the performance is then back again on 5.6.0 levels (even a little bit better). I hope that resolves the 'new' issues. What we do about jazzy and if we can backport those fixes also to humble, is a bit open.

Cheers,
Andreas

lianghongzhuo, 2025-11-06:
> > We tested the gravity_compensation_example on Jazzy with 5.8.2 image. This gave no problems.
> > We only face the error when using the joint_trajectory_controller.
> 
> If you are on jazzy can you enable the introspection tools to diagnose the controllers especially the joint trajectory controller update function? You can visualize the introspection tools on plotjuggler. Can you analyze if there is sth irregular e.g. some spikes in the timings etc. Also please make sure you build everything with the Release flag

@BarisYazici  Could you give an introduction on how to enable this test? How to enable the introspection tools ?

lianghongzhuo, 2025-11-07:
> > We also still experience the communication constraints violations error. We use 5.8.2 firmware. We have tested with both Jazzy and Humble using the Docker container installs provided in the README of both branches.
> > We can successfully use different [example controllers](https://frankarobotics.github.io/docs/franka_ros2/franka_example_controllers/doc/index.html), but we always experience the error when using a joint_trajectory_controller. The easiest way to test this for us is by starting the [moveit example](https://frankarobotics.github.io/docs/franka_ros2/franka_fr3_moveit_config/doc/index.html):
> > `ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=<fci-ip>`
> > Even when we only load and activate the joint_trajectory_controller (without sending commands), the error sometimes occurs. It is very inconsistent. Most of the time, the error occurs within a few seconds after activating the joint_trajectory_controller. Sometimes we can do a few movements, but we were never able to last more than a minute before failing again because of the communication constraints violations.
> 
> [@Jelmerdw](https://github.com/Jelmerdw) Have you tried to adapt the franka_fr3_moveit_config/config/fr3_ros_controllers.yaml ? Just add "enable_overrun: false" as an additional option under the ros_parameters like [@BarisYazici](https://github.com/BarisYazici) did for franka_bringup with commit [99e9a64](https://github.com/frankarobotics/franka_ros2/commit/99e9a6432bccb22eafb80c15b1c917f9325edae4) to actually apply the hotfix.

That is exactly what I misunderstood with @BarisYazici , I thought the by defailt config is the hot fix, and did not add "enable_overrun: false" on my ros2 controller yaml file. I would be exicited to test if this works/fix the 
communication_constraints_violation problem.

I will report once I tested this.

BarisYazici, 2025-11-08:
`@Jelmerdw Have you tried to adapt the franka_fr3_moveit_config/config/fr3_ros_controllers.yaml ? Just add "enable_overrun: false" as an additional option under the ros_parameters like @BarisYazici did for franka_bringup with commit https://github.com/frankarobotics/franka_ros2/commit/99e9a6432bccb22eafb80c15b1c917f9325edae4 to actually apply the hotfix.`

that's a good catch. We actually forgot to add the enable_overrun to false for the moveit examples. Please add enable_overrun: false parameter to this controller yaml https://github.com/frankarobotics/franka_ros2/blob/humble/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml
We will fix it as soon as possible in the main branch

mrceki, 2025-11-10:
Hi guys,
I've upgraded my robot to version 5.9.0 and gave it a go with humble, and unfortunately it did fail again with gravity compensation error. Then I rebuild the docker container with jazzy version and it seems like it did the job and I did not get 'communication_constraints_violation' for a while :)

Jelmerdw, 2025-11-12:
@magnusdierking @BarisYazici 

Adding the `enable_overrun: false` option indeed solves the problem for us!
We tested again on Jazzy, 5.8.2 and can we can use the joint_trajectory_controller without problems now by adding the option.
Thanks a lot for the help!